NVIDIAGameWorks/PhysX #110: PxArticulationJointReducedCoordinate current drive force

nov-ta, 2019-04-24:
I'm currently modelling a robotic arm with a grabber, and I'm trying to figure out the actual force a joint drive is currently applying to an articulation joint, but I'm having no luck so far. 

I've been looking into:
`PxArticulationReducedCoordinate::computeGeneralizedGravityForce 
computeJointForce 
computeGeneralizedExternalForce 
computeLambda
computeCoefficientMatrix`

I can't seem to get any results that will include forces that counter objects resting on them. Forces that counter gravity seem sensible to me, and also "raw" joint forces that are derived from joint acceleration, but I can't seem to find something that will allow me to sum up all forces so I can get the total drive force that include force needed to counter objects that are being held up by the articulation. 

The forces reported seem identical whether or not there is something resting on or being held up by the articulation. The docs state something about PxContactJoint and loop joints that seem to be used in lambda and coefficient matrix calculation, but it's not clear to me how to use them.

Any pointers would be greatly appreciated!

xgerrmann, 2019-04-26:
I think due the nature of the implementation you cannot get joint (and also drive) forces and moments.

This is what the [documentation](http://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/Articulations.html?highlight=articulation) says about it:
> Articulation joints are not breakable, and it is not possible to retrieve the constraint force applied at the joint.

However I also ran into the following in the section regarding inverse dynamics:
> articulation->computeJointForce(*cache);

So it seems impossible to directly query a joint (possibly also a drive) force, but maybe it is feasible through the inverse joint Dynamics?

Good luck ;)

nov-ta, 2019-04-26:
Hello, thanks for the input! I've been trying **computeJointForce** among other functions on the articulation cache, but none of them seem to return values that are affected by "external" objects, that is, they seem sensible in reporting force that is required to counteract gravity to keep the current pose, but if I grab objects or put objects on top of the articulation, the reported forces seem unaffected.

I'm currently trying to dig a bit around in DyFeatherstoneArticulation to see if I can make sense of the forces in there, but so far I don't have any solution, tbh much of this is pretty difficult for me to grasp. The external forces do seem to be affecting the articulation, as I can tweak the max joint drive force so external objects can make the drives fail to reach their target, but I'm not sure how to get that info back out into user code.

xgerrmann, 2019-04-26:
I also ran into this forum post: https://devtalk.nvidia.com/default/topic/1037071/physics-modeling/pxd6joint-drive-force/post/5268847/#5268847

The post is about the forces from a PxD6 driven joint. It might also apply to the articulation.

nov-ta, 2019-04-26:
Thanks, will look into that! Appreciate the effort!

kstorey-nvidia, 2019-04-26:
We don't currently expose the drive forces in articulation drives. I don't actually know the reason why the design decision not to expose joint drive forces was made. I think we could report at least the drive forces around each degree of freedom. I'll add this to the list of feature requests and it should hopefully get looked into in the near future. The drive forces are stored internally in ArticulationInternalConstraint::driveForce, which you should be able to find updated in the solveInternalConstraints method in DyFeatherstoneArticulation.cpp.

You can use inverse dynamics to compute the joint torques required to counteract gravity, external forces, internal forces (Coriolis/centrifugal forces). 

You can also use inverse dynamics to compute forces to counteract contacts but this is quite involved and it is a non-trivial problem to solve because applying counteracting forces will often change the forces that a contact applies. 

Inverse dynamics itself does not know about the set of contacts that are affecting the simulation on a given frame. However, you can define these contacts by adding loop PxContactJoints and using computeLambda to compute the joint forces to counteract these contacts. This is an iterative process that attempts to converge on a global solution; if no such solution can be found, the current 'best' solution is returned.

nov-ta, 2019-04-26:
Good to know, I've been looking inside solveInternalConstrains, I'll see if I can get the values back out to where I need  them.

I did briefly look into PxContactJoints and computeLambda, but it wasn't clear to me how to properly set that up. As far as I could see from my testing these contact joints were not automatically registered with the articulation, so I put that track on hold for now as I was hoping the forces would be available more easily.

A simple way to get out the current drive forces for the unlocked degrees of freedom would be a very welcome addition :)

Thanks for clearing this up!

closed 2019-04-26 by nov-ta

kstorey-nvidia, 2019-04-26:
You are welcome.

You are right that PxContactJoint is not automatically registered. Inverse dynamics functions currently require the user to define any forces and constraints on the simulation. There's no collision detection performed and they don't pull any contact information out of the previous frame.

Depending on what you want to do, there are sometimes very simple solutions that don't require much hoop jumping. For example, if you want to figure out how much joint torque required to hold an object, you could potentially just add that much mass to the end effector, or add an additional link to the configuration to represent the object you want to lift (connected via a fixed joint), and compute the forces required to counteract gravity etc.. This would tell you the joint torques required to hold, say, a robot arm up while grasping an object of a given mass. It would not, however, tell you the joint torques required to make a gripper grasp the object.