NVIDIAGameWorks/PhysX #115: How to link and include PhysX 4.0 Extensions Library?

Fr33ze, 2019-04-29:
I am including the PhysX 4.0 SDK into my C++ opengl project, but I cannot use anything from the PhysX Extensions library like `PxDefaultErrorCallback` or `PxDefaultMemoryOutputStream` because I need to include the Extension library first. The problem is I cannot find the `PhysXExtensions32.lib` and its DLL in any folder of the PhysX SDK that I have compiled myself.

I could find a static library called `PhysXExtensions_static_32.lib`, but when linking against this library I get this error:
`error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MDd_DynamicDebug' in file.obj.`

Is there a way I can get the Extension libaray as `Multithreaded-Debug-DLL (/MDd)` instead of a `Multithreaded (/MT)`? I can only use /MDd, because I use other libraries like glew and glfw and I get another error when I switch to /MT.

I link against these libraries:
- PhysX_32.lib
- PhysXCommon_32.lib
- PhysXCooking_32.lib
- PhysXFoundation_32.lib

I redistribute these DLL files:
- PhysX_32.dll
- PhysXCommon_32.dll
- PhysXCooking_32.dll
- PhysXFoundation_32.dll
- PhysXGpu_32.dll

AlesBorovicka, 2019-04-29:
Hi,
 yes, its described in [https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/documentation/PhysXGuide/Manual/BuildingWithPhysX.html](url)

You can toggle static/dynamic runtime with a NV_USE_STATIC_WINCRT  switch, this switch is in buildtools/presets/public/"your preset that you uses for compilation like vc12win32"

closed 2019-04-30 by Fr33ze

philipswan, 2019-08-21:
The link in the response is broken

Fr33ze, 2019-08-22:
Hi,
its all described here (same doc that the dead link above refers to):
https://gameworksdocs.nvidia.com/PhysX/4.0/documentation/PhysXGuide/Manual/BuildingWithPhysX.html

If you still have problems, let me know.

Farost, 2020-10-07:
Hi,
I've checked the link, copied all *.dlls,*.libs, and includes to my project folder, configured my CMakeLists.txt, adding "include_directories()" with physx/include, "add_library()" with properties of include, PhysX_64.dll, and PhysX_64.lib, and linked the library to my project. 

It works well, and I have no issues with using upper *.h files and the packages which have their dll files (like PhysXFoundation), but I can't work with Extensions, because linkage throws "unresolved external symbol" when I'm trying to use something (i.e. PxDefaultErrorCallback), and I can't undestand how to provide it the declaration of the classes in the library since I have only PhysXExtensions_static_64.lib and no dlls for it. 

I've tried turning off NV_USE_STATIC_WINCRT, but it generates the same _static_64.lib files, which don't help. Am I doing something wrong?

DanteMarshal, 2021-04-22:
I'm having the same issue using static build. I checked all the requirements and I have them installed.
I'm building on Debug configuration and including libraries using the $(Configuration) variable in VS2019 so there shouldn't be a mismatch there.
The linker is successfully finding .lib files and searching them but it can't find the symbols inside them. At the end of the linker's verbose log they're listed under "Unused Libraries" section.
Here's the complete verbose log of the linker :
[Linker.log](https://github.com/NVIDIAGameWorks/PhysX/files/6356319/Linker.log)
And here's my build preset :
```
<?xml version="1.0" encoding="utf-8"?>
<preset name="vc16win64" comment="VC16 Win64 PhysX general settings">
  <platform targetPlatform="win64" compiler="vc16" />
  <CMakeSwitches>
    <cmakeSwitch name="PX_BUILDSNIPPETS" value="False" comment="Generate the snippets" />
    <cmakeSwitch name="PX_BUILDPUBLICSAMPLES" value="False" comment="Generate the samples projects" />
    <cmakeSwitch name="PX_GENERATE_STATIC_LIBRARIES" value="True" comment="Generate static libraries" />
    <cmakeSwitch name="NV_USE_STATIC_WINCRT" value="False" comment="Use the statically linked windows CRT" />
    <cmakeSwitch name="NV_USE_DEBUG_WINCRT" value="True" comment="Use the debug version of the CRT" />
    <cmakeSwitch name="PX_FLOAT_POINT_PRECISE_MATH" value="False" comment="Float point precise math" />
  </CMakeSwitches>
  <CMakeParams>
    <cmakeParam name="CMAKE_INSTALL_PREFIX" value="install/vc15win64/PhysX" comment="Install path relative to PhysX SDK root" />
  </CMakeParams>
</preset>
```

Can anyone help here ?
I'm kinda stuck

SudharsanZen, 2021-05-31:
> I am including the PhysX 4.0 SDK into my C++ opengl project, but I cannot use anything from the PhysX Extensions library like `PxDefaultErrorCallback` or `PxDefaultMemoryOutputStream` because I need [...]

I have the same problem and i set PX_GENERATE_STATIC_LIBRARIES to false as described in the documentation.
but i still get the same error. Can you please say what exactly you changed to make it work?

AlesBorovicka, 2021-05-31:
In the preset file: buidtools/presets you can define what runtime should be used:
`<cmakeSwitch name="NV_USE_STATIC_WINCRT" value="True" comment="Use the statically linked windows CRT" />`

Pacheco95, 2022-10-17:
I'm facing the same problem with static linking.

I'm using this custom preset:

```xml
<?xml version="1.0" encoding="utf-8"?>
<preset name="vc16win32" comment="VC16 Win32 PhysX general settings">
  <platform targetPlatform="win32" compiler="vc16" />
  <CMakeSwitches>
    <cmakeSwitch name="PX_BUILDSNIPPETS" value="False" comment="Generate the snippets" />
    <cmakeSwitch name="PX_BUILDPUBLICSAMPLES" value="False" comment="Generate the samples projects" />
    <cmakeSwitch name="PX_GENERATE_STATIC_LIBRARIES" value="True" comment="Generate static libraries" />
    <cmakeSwitch name="NV_USE_STATIC_WINCRT" value="True" comment="Use the statically linked windows CRT" />
    <cmakeSwitch name="NV_USE_DEBUG_WINCRT" value="True" comment="Use the debug version of the CRT" />
    <cmakeSwitch name="PX_FLOAT_POINT_PRECISE_MATH" value="False" comment="Float point precise math" />
  </CMakeSwitches>
  <CMakeParams>
    <cmakeParam name="CMAKE_INSTALL_PREFIX" value="install/vc15win32/PhysX" comment="Install path relative to PhysX SDK root" />
  </CMakeParams>
</preset>
```

The visual studio solution was built sucessfully and the library build and installation too.

I'm using this simple snippet from [https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/Startup.html](https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/Startup.html)

```c++
#define _SILENCE_CXX20_CISO646_REMOVED_WARNING

#include "PxPhysicsAPI.h"

using namespace physx;

static PxDefaultErrorCallback gDefaultErrorCallback;
static PxDefaultAllocator gDefaultAllocatorCallback;

auto mFoundation = PxCreateFoundation(PX_PHYSICS_VERSION, gDefaultAllocatorCallback, gDefaultErrorCallback);

int main() {
    if (!mFoundation) {
        return EXIT_FAILURE;
    }
    return EXIT_SUCCESS;
}
```

Build error:

```
Rebuild started...
1>------ Rebuild All started: Project: SFML, Configuration: Debug Win32 ------
1>Game.cpp
1>main.cpp
1>Generating Code...
1>PhysXExtensions_static_32.lib(ExtDefaultErrorCallback.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Game.obj
1>PhysXExtensions_static_32.lib(ExtDefaultErrorCallback.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MDd_DynamicDebug' in Game.obj
1>PhysXFoundation_static_32.lib(PsWindowsPrintString.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Game.obj
1>PhysXFoundation_static_32.lib(PsWindowsPrintString.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MDd_DynamicDebug' in Game.obj
1>PhysXFoundation_static_32.lib(PsWindowsThread.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Game.obj
1>PhysXFoundation_static_32.lib(PsWindowsThread.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MDd_DynamicDebug' in Game.obj
1>PhysXFoundation_static_32.lib(PsFoundation.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Game.obj
1>PhysXFoundation_static_32.lib(PsFoundation.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MDd_DynamicDebug' in Game.obj
1>PhysXFoundation_static_32.lib(PsAllocator.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Game.obj
1>PhysXFoundation_static_32.lib(PsAllocator.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MDd_DynamicDebug' in Game.obj
1>PhysXFoundation_static_32.lib(PsWindowsMutex.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Game.obj
1>PhysXFoundation_static_32.lib(PsWindowsMutex.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MDd_DynamicDebug' in Game.obj
1>PhysXFoundation_static_32.lib(PsString.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in Game.obj
1>PhysXFoundation_static_32.lib(PsString.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MDd_DynamicDebug' in Game.obj
1>LINK : warning LNK4098: defaultlib 'LIBCMT' conflicts with use of other libs; use /NODEFAULTLIB:library
1>LINK : warning LNK4217: symbol '_PxCreateFoundation' defined in 'PhysXFoundation_static_32.lib(PsFoundation.obj)' is imported by 'main.obj' in function '"void __cdecl `dynamic initializer for 'mFoundation''(void)" (??__EmFoundation@@YAXXZ)'
[... 139 lines cut ...]
    </ClCompile>
    <Link>
      <SubSystem>Console</SubSystem>
      <EnableCOMDATFolding>true</EnableCOMDATFolding>
      <OptimizeReferences>true</OptimizeReferences>
      <GenerateDebugInformation>true</GenerateDebugInformation>
    </Link>
  </ItemDefinitionGroup>
  <ItemGroup>
    <ClCompile Include="Game.cpp" />
    <ClCompile Include="main.cpp" />
  </ItemGroup>
  <ItemGroup>
    <ClInclude Include="constants.h" />
    <ClInclude Include="Game.h" />
    <ClInclude Include="utils.h" />
    <ClInclude Include="Vec2.h" />
  </ItemGroup>
  <Import Project="$(VCTargetsPath)\Microsoft.Cpp.targets" />
  <ImportGroup Label="ExtensionTargets">
  </ImportGroup>
</Project>
```

</details>