NVIDIAGameWorks/PhysX #120: Setting PxRigidDynamicLockFlag on rigidbody results in broken collisions

tysoni, 2019-05-10:
I have a scene with a large rigidbody box and some smaller rigidbodies that land on it. 

When I set the box to kinematic, it works as intended (smaller RBs land on it and collide as expected).

When I keep the box dynamic, but lock all of its pos/rot axis by setting PxRigidDynamicLockFlag on them, the smaller RBs fall through it.

My intuition is that lock flags don't actually keep RBs static within the sim, but instead simply restore position/rotation values at the end of the time step, which would explain the tunneling I'm experiencing in my sim, however, I'm asking here to get an official explanation.

Here is a video showing the issue. On the left, the base collider is dynamic with all axis for pos/rot locked. On the right it is set to kinematic.

[image]

kstorey-nvidia, 2019-05-10:
Your intuition is correct. The locked axes do a little bit more than just removing motion around the disallowed axes (there's some pre-post solver velocity re-projection and some inertia alignment/adjustments), but this is not enough to replace a static rigid body. This feature's main purpose is to make it relatively cheap to emulate something like a 2D simulation using PhysX, where you disallow rotation/translation around/along one or more axes. An alternative way to achieve this would be to use constraints, but this will be more expensive. 

The main reason why your example doesn't behave as a static body is that we do not support per-axis invMass properties for rigid bodies, which would be required to make the body appear to have infinite mass along some axes and finite mass along others.

tysoni, 2019-05-21:
Thank you for the explanation!

closed 2019-05-21 by tysoni