NVIDIAGameWorks/PhysX #124: Physx 4.1 - d6 joint limit bug

iaiotom, 2019-05-20:


iaiotom, 2019-05-21:
I've found that inside the static PxU32 D6JointSolverPrep(…) function the PxQuat swing variable is computed without consider the constraint rotation and since the angle limit check is using this quaternion to check the boundaries I get the problem I show before.

kstorey-nvidia, 2019-05-21:
Are you sure that it is computed without considering the constraint rotation? Swing/twist are decomposed from cB2cA transform, which is computed from cA2w and cB2w transforms, which are the joint frames transformed into world space. These are computed as a byproduct of the ConstraintHelper constructor. 

Are you certain that the joint frames have been correctly configured and exported from Maya?

closed 2019-05-28 by iaiotom