NVIDIAGameWorks/PhysX #126: Understanding the joint configuration format of articulated bodies in PxArticulationCache

lasagnaphil, 2019-05-27:
Right now the thing I ultimately want to do is to directly convert the joint rotations for an articulated body to quaternions, so I can use the state of the articulated body for other purposes (such as reinforcement learning). I was trying to use the PxArticulationCache to access the joint configurations, but am having trouble doing so because I have no idea about the data format of the raw array (comprised of PxReal values). It doesn't seem to be in quaternions, because the 
DOF of the rotations from ``link->getInboundJointDof()`` were all 3. The closest I could find for how rotations are expressed internally was in the description for the D6Joint:

"Rotational degrees of freedom are partitioned as twist (around the X-axis of actor0's constraint frame) and swing (around the Y- and Z- axes). Different effects are achieved by unlocking various combinations of twist and swing."

So is the format (twist, swing1, swing2) used for representing joint rotations in the articulation cache? If so, is the twist rotation applied first, and then the rotation swing1 around the Y axis and the rotation swing2 around the Z axis are applied consecutively? Or am i getting the order of rotations backwards?  Or is it the whole thing in exponential coordinates? (Because that is what I think is used widely in robotics simulations nowadays.) A more detailed explanation of the data format would help a lot.

Best regards,
Phil

nov-ta, 2019-05-31:
Hello,

I'm a bit surprised getting to these values require a bit of error-prone arithmetic, but I've been using the indexing explained under the paragraph:

**'The location of the entries corresponding to a given link can be found by summing the number of degrees of freedom of all preceding links'**

https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/Articulations.html#pxarticulationcache

I think the DoFs are laid out in the array sequentially from "dofStarts" based on what DoFs you have for a link. In my case, I mostly have one degree of freedom for each link, so I usually just do for example something along the lines of 
`jointPos = cache->jointPosition[dofStarts[link->getLinkIndex()]];`

If you have more DoFs, as far as I can tell they're laid out in the order the PxArticulationAxis enum is defined, so for example if you have a joint with eSwing and eY, eSwing should be 
`cache->jointPosition[dofStarts[link->getLinkIndex()]] `
and eY should be 
`cache->jointPosition[dofStarts[link->getLinkIndex()] + 1]`

"jointPos" in this case is an angle around the given axis for revolute joints, and an offset along the given axis for prismatic joints. I'm using this code to get the local offsets of a joint, but keep in mind this is meant for joints with just one DoF:

```
PxArticulationLink& link = mJoint->getChildArticulationLink();

PxU32 linkIndex = link.getLinkIndex();
PxU32 dofStart = mLinkDofStarts[linkIndex];

float currentPosition = mArticulationCache->jointPosition[dofStart];

if (axis == PxArticulationAxis::eSWING2)
{
	return PxTransform(PxQuat(currentPosition, PxVec3(0.0f, 0.0f, 1.0f)));
}
	
if (axis ==  PxArticulationAxis::eY)
{
	return PxTransform(PxVec3(0.0f, currentPosition, 0.0f));
}

```	

I've been trying to dig around inside the source to see how these things fit together, but I've already gotten this wrong more than once, so please let me know if this doesn't seem to make sense :S **FeatherstoneArticulation::setupInternalConstraintsRecursive** have given me a couple of enlightening moments - if I got it right.

lasagnaphil, 2019-06-18:
I've finally figured out the format of the cache after fiddling with the engine for some time. 

The positions of the articulation joints are in Euler angles (ZYX format, Tait-bryan), so the order of rotations applied is X -> Y -> Z. (So the equation for finding the rotation would be q = q_z * q_y * q_x.) The rotation around X represents the twist of the rotation, and the rotation around Y and Z represents the swing of the rotation. (So 1 twist DOF + 2 swing DOF). One thing to be careful is that the X, Y and Z axes of the Euler angles represent the axes of the joint's local frame, not the global frame (so when the rotation around X happens, the object is only twisted by itself and there is no swing). Therefore to decompose your rotation into twist-swing coordinates for the cache, you need to do a coordinate transformation first.

Maybe this was obvious for many people already using this engine, but it would be great if there was some brief note about it in the docs for beginners...

closed 2019-06-26 by lasagnaphil