NVIDIAGameWorks/PhysX #127: PhysX 4.1 crashing on release.

TomerZeitune, 2019-05-29:
I updated my engine to PhysX 4.1 from PhysX 4.0. (The code using physX didn't change).
When I run on debug everything is working normally. When I run on release physX crash after about a second. The simulation is working as it should at that second (As far as I can tell). 

Exception details:
> Exception thrown at 0x00007FF95C582D7B (PhysXCommon_64.dll) in Application.exe: 0xC0000005: Access violation reading location 0xFFFFFFFFFFFFFFFF. occurred
its at line 58 at the file PsVecTransform.h 
here's the function throwing it:
```
PX_FORCE_INLINE PsTransformV(const Vec3VArg p0 = V3Zero(), const QuatVArg q0 = QuatIdentity()) : q(q0), p(p0)
{
	PX_ASSERT(isSaneQuatV(q0));
}

```
*Note all the scene is rendering in purple (when on release mode only) since physX has nothing to do with rendering I assume this is a bug on my side. 

I use the release libs and dlls for release and the debug libs and dlls for debug.
My preset is vs14win64 but I changed the cmake switch "NV_USE_STATIC_WINCRT" to "False"
My platform is windows 10. Running a 64 bit version of physx.
Using physX on the CPU. 
I use these libs
                PhysXCommon_64.lib
		PhysX_64.lib
		PhysXFoundation_64.lib
		PhysXExtensions_static_64.lib
		PhysXCooking_64.lib

Not sure if physX is logging errors since my logging is totally off on release mode. I can try to enable it somehow if needed.

kstorey-nvidia, 2019-05-29:
Are there any errors logged by the debug build? Debug and checked PhysX both validate user input and throw away illegal parameters while logging warnings. All this validation is removed in release build.

Could you try linking against the checked build config to see if the issue repeats or perhaps there are errors logged by PhysX?

If the above doesn't help identify the issue, could you please provide a callstack?

TomerZeitune, 2019-05-29:
No errors logged by the debug build. The debug build is working perfectly just as PhysX 4.0 did. 

The checked build is, unfortunately, using the  MdRelease runtime. while the debug config of my engine uses the MdDebug runtime. 

So I can't link PhysX checked to my debug config. But I can link physX checked to the release config of my engine. It crashes too.

> Exception thrown at 0x00007FF950FD3E6B (PhysXCommon_64.dll) in Application.exe: 0xC0000005: Access violation reading location 0xFFFFFFFFFFFFFFFF.

Added a small check to my logging system no errors or warnings are logged at any configuration of physX or my engine.

Changed the Convex Mesh collider(PxConvexMesh) to a box collider (PxBoxGeometry). Every config is working perfectly. The issue seems to be exclusive to PxConvexMesh collision.

cmp-, 2019-05-29:
Are you compiling with a Visual Studio 2019 version earlier than 16.1 by any chance? There was a bug that optimized out unaligned loads, which generates exactly that error (Access violation reading location 0xFFFFFFFFFFFFFFFF).

TomerZeitune, 2019-05-29:
@cmp Yes. I am compiling with Visual Studio 2019 16.0.3. I'll update Visual Studio and recompile PhysX.

TomerZeitune, 2019-05-29:
Updating Visual Studio 2019 to version 16.1.1 and recompiling PhysX release solved the issue.

closed 2019-05-29 by TomerZeitune