NVIDIAGameWorks/PhysX #128: Performance of the function PxRigidActor::setGlobalPose

EverThunder, 2019-05-29:
Hi,
In the function comments(PxRigidActor.h)
"This method is mainly for dynamic rigid bodies (see #PxRigidDynamic). Calling this method on static actors is 	likely to result in a performance penalty, since internal optimization structures for static actors may need to be 	recomputed. ...."
virtual void setGlobalPose(const PxTransform& pose, bool autowake = true) = 0;

I took one test in a empty scene.
When moved nine hundred objects of PxRigidStatic, the fps was sixty on my computer.
When moved nine hundred objects of kinematic PxRigidDynamic, the fps was fifty.
The results differ from expectations.

kstorey-nvidia, 2019-05-29:
This description is a little old/stale. Performance isn't significantly different between moving statics or kinematics. The overhead was mainly observed when updating the SQ pruners, which now use the same data structure for dynamics/kinematics and statics by default.

EverThunder, 2019-05-29:
Forget to say, my physics version is 3.3.3. Is there any performace difference between scene query dynamic pruners or static pruners?

AlesBorovicka, 2019-05-29:
There is no different in 3.3. The static pruner is set to use dynamic AABB tree too:
	staticStructure						(PxPruningStructure::eDYNAMIC_AABB_TREE),
	dynamicStructure					(PxPruningStructure::eDYNAMIC_AABB_TREE),
With these settings, there is no penalty in the SQ pruners, when next query is done. If eSTATIC_AABB_TREE would have been used, then the tree would have been rebuilt during first sq flush update and thats expensive. But thats not the case anymore with default settings.