NVIDIAGameWorks/PhysX #129: PhysX GPU SharedMemory Exception

syslot, 2019-05-29:
Hi, 

Based on the bullet3's source, I load urdf in PhysX. Referring to the snippethellogrb, I load urdf into PhysX. When I call PxScene::simulate, libPhysXGpu.so throws an exception  "allocAligned out of sharedMemory" with GPU,  while source works well when only use CPU.  

Can anyone tell me what causes this problem?

syslot, 2019-05-29:
Here is the stack.

#0  0x00007ffff0e3ea23 in physx::PxgCudaSolverCore::syncDmaBack(unsigned int&) () from /home/syslot/physx_gpu/libPhysXGpu_64.so
#1  0x00007ffff0e2739e in physx::PxgDynamicsContext::doPostSolveTask(physx::PxBaseTask*) () from /home/syslot/physx_gpu/libPhysXGpu_64.so
#2  0x0000555555e06585 in physx::Cm::Task::run (this=0x555557652e70) at /home/syslot/DevSpace/bullet3/src/PhysX/physx/source/common/src/CmTask.h:66
#3  0x0000555555e70964 in physx::Ext::DefaultCpuDispatcher::runTask (this=0x555556a1dca0, task=warning: RTTI symbol not found for class 'physx::PxgPostSolveTask'
...)
    at /home/syslot/DevSpace/bullet3/src/PhysX/physx/source/physxextensions/src/ExtDefaultCpuDispatcher.h:97
#4  0x0000555555f389d8 in physx::Ext::DefaultCpuDispatcher::submitTask (this=0x555556a1dca0, task=warning: RTTI symbol not found for class 'physx::PxgPostSolveTask'
...)
    at /home/syslot/DevSpace/bullet3/src/PhysX/physx/source/physxextensions/src/ExtDefaultCpuDispatcher.cpp:129
#5  0x00005555560c6776 in physx::PxTaskMgr::decrReference (this=0x555556f11420, lighttask=warning: RTTI symbol not found for class 'physx::PxgPostSolveTask'
...)
    at /home/syslot/DevSpace/bullet3/src/PhysX/physx/source/task/src/TaskManager.cpp:174
#6  0x0000555555e00f2a in physx::PxLightCpuTask::removeReference (this=0x555557652e70) at /home/syslot/DevSpace/bullet3/src/PhysX/physx/include/task/PxTask.h:277
#7  0x00005555562ee0bc in physx::Sc::Scene::updateSimulationController (this=0x555556f01820, continuation=0x555556f02d40)
    at /home/syslot/DevSpace/bullet3/src/PhysX/physx/source/simulationcontroller/src/ScScene.cpp:2862
#8  0x000055555629c385 in physx::Cm::DelegateTask<physx::Sc::Scene, &physx::Sc::Scene::updateSimulationController>::runInternal (this=0x555556f02de8)
    at /home/syslot/DevSpace/bullet3/src/PhysX/physx/source/common/src/CmTask.h:100
#9  0x0000555555e06585 in physx::Cm::Task::run (this=0x555556f02de8) at /home/syslot/DevSpace/bullet3/src/PhysX/physx/source/common/src/CmTask.h:66
#10 0x0000555555e70964 in physx::Ext::DefaultCpuDispatcher::runTask (this=0x555556a1dca0, task=...)
    at /home/syslot/DevSpace/bullet3/src/PhysX/physx/source/physxextensions/src/ExtDefaultCpuDispatcher.h:97
#11 0x0000555555f389d8 in physx::Ext::DefaultCpuDispatcher::submitTask (this=0x555556a1dca0, task=...)
    at /home/syslot/DevSpace/bullet3/src/PhysX/physx/source/physxextensions/src/ExtDefaultCpuDispatcher.cpp:129
#12 0x00005555560c6776 in physx::PxTaskMgr::decrReference (this=0x555556f11420, lighttask=...)
    at /home/syslot/DevSpace/bullet3/src/PhysX/physx/source/task/src/TaskManager.cpp:174
#13 0x0000555555e00f2a in physx::PxLightCpuTask::removeReference (this=0x555556f02de8) at /home/syslot/DevSpace/bullet3/src/PhysX/physx/include/task/PxTask.h:277
#14 0x0000555555e00fb0 in physx::PxLightCpuTask::release (this=0x555556f02e20) at /home/syslot/DevSpace/bullet3/src/PhysX/physx/include/task/PxTask.h:304
#15 0x0000555555f389ef in physx::Ext::DefaultCpuDispatcher::submitTask (this=0x555556a1dca0, task=...)
    at /home/syslot/DevSpace/bullet3/src/PhysX/physx/source/physxextensions/src/ExtDefaultCpuDispatcher.cpp:130
#16 0x00005555560c6776 in physx::PxTaskMgr::decrReference (this=0x555556f11420, lighttask=...)
    at /home/syslot/DevSpace/bullet3/src/PhysX/physx/source/task/src/TaskManager.cpp:174
#17 0x0000555555e00f2a in physx::PxLightCpuTask::removeReference (this=0x555556f02e20) at /home/syslot/DevSpace/bullet3/src/PhysX/physx/include/task/PxTask.h:277
#18 0x0000555555e00fb0 in physx::PxLightCpuTask::release (this=0x555556f02db0) at /home/syslot/DevSpace/bullet3/src/PhysX/physx/include/task/PxTask.h:304
#19 0x0000555555f389ef in physx::Ext::DefaultCpuDispatcher::submitTask (this=0x555556a1dca0, task=...)
    at /home/syslot/DevSpace/bullet3/src/PhysX/physx/source/physxextensions/src/ExtDefaultCpuDispatcher.cpp:130
#20 0x00005555560c6776 in physx::PxTaskMgr::decrReference (this=0x555556f11420, lighttask=...)
    at /home/syslot/DevSpace/bullet3/src/PhysX/physx/source/task/src/TaskManager.cpp:174
#21 0x0000555555e00f2a in physx::PxLightCpuTask::removeReference (this=0x555556f02db0) at /home/syslot/DevSpace/bullet3/src/PhysX/physx/include/task/PxTask.h:277
#22 0x0000555555e00fb0 in physx::PxLightCpuTask::release (this=0x555556f02d78) at /home/syslot/DevSpace/bullet3/src/PhysX/physx/include/task/PxTask.h:304
#23 0x0000555555f389ef in physx::Ext::DefaultCpuDispatcher::submitTask (this=0x555556a1dca0, task=...)
    at /home/syslot/DevSpace/bullet3/src/PhysX/physx/source/physxextensions/src/ExtDefaultCpuDispatcher.cpp:130
#24 0x00005555560c6776 in physx::PxTaskMgr::decrReference (this=0x555556f11420, lighttask=...) at /home/syslot/DevSpace/bullet3/src/PhysX/physx/source/task/src/TaskManager.cpp:174
#25 0x0000555555e00f2a in physx::PxLightCpuTask::removeReference (this=0x555556f02d78) at /home/syslot/DevSpace/bullet3/src/PhysX/physx/include/task/PxTask.h:277
#26 0x0000555555e00fb0 in physx::PxLightCpuTask::release (this=0x555556f03050) at /home/syslot/DevSpace/bullet3/src/PhysX/physx/include/task/PxTask.h:304
#27 0x0000555555f389ef in physx::Ext::DefaultCpuDispatcher::submitTask (this=0x555556a1dca0, task=...)
    at /home/syslot/DevSpace/bullet3/src/PhysX/physx/source/physxextensions/src/ExtDefaultCpuDispatcher.cpp:130
#28 0x00005555560c6776 in physx::PxTaskMgr::decrReference (this=0x555556f11420, lighttask=...) at /home/syslot/DevSpace/bullet3/src/PhysX/physx/source/task/src/TaskManager.cpp:174
[... cut ...]
#66 0x0000555555cf7748 in b3SubmitClientCommand (physClient=0x555556a17fa0, commandHandle=0x7ffff3921038)
    at /home/syslot/DevSpace/bullet3/examples/SharedMemory/PhysicsClientC_API.cpp:2424
#67 0x0000555555cf7862 in b3SubmitClientCommandAndWaitStatus (physClient=0x555556a17fa0, commandHandle=0x7ffff3921038)
    at /home/syslot/DevSpace/bullet3/examples/SharedMemory/PhysicsClientC_API.cpp:2448
#68 0x0000555555c22f34 in main () at /home/syslot/DevSpace/bullet3/examples/PhysxDemo/PhysxDemoMain.cpp:106

kstorey-nvidia, 2019-05-29:
Thanks. This is a known issue and we already have a fix for it. It should be included in the next PhysX release.

syslot, 2019-05-29:
> Thanks. This is a known issue and we already have a fix for it. It should be included in the next PhysX release.

Thanks for your replying. Can I know the release time for the reason that I need to use this function to continue my project :)

syslot, 2019-06-03:
> Thanks. This is a known issue and we already have a fix for it. It should be included in the next PhysX release.

Hi kstorey,

I still wait for your reply about your next release date. If you are not sure about the release time, can you provide a night build version for me to test?

closed 2019-06-05 by syslot

phcerdan, 2019-11-05:
I would suggest to enclose the stack reported in the comment: https://github.com/NVIDIAGameWorks/PhysX/issues/129#issuecomment-496959725 with the code ticks: \``` . Right now this issue is referencing almost all the other issues due to the `#x` format in the stack report.

syslot, 2019-11-07:
Thanks for your advise and updated.