NVIDIAGameWorks/PhysX #132: D6 Joint unexpected behaviour of SwingY, swingZ and Twist

xgerrmann, 2019-06-04:
Hello, 

Let me start by describing my scene as shown in the image below. I have a reduced coordinate articulation in my world. The articulation root is on top. It is attached to a PxRigidDynamic on the bottom through a D6 joint.

Gravity is turned off and no exernal forces are applied, therefore the actors all go to sleep at some point.

What strikes me is the values I get from the D6 joint. After each simulation step I get the following values for Twist, SwingY and Swingz (see below). This behavior seems very erratic since the angles are all over the place while the scene is quite static. Can anyone explain this behavior to me?
```
-1.96559, 2.77592, 2.02578
-1.5708, 2.46192, 2.46192
4.71239, -2.46192, -2.46192
0, 0, 3.14159
3.14159, -3.14159, 1.19209e-07
-1.5708, 2.46192, 2.46192
-1.96559, 2.77592, 2.02578
-1.5708, 2.46192, 2.46192
4.71239, -2.46192, -2.46192
0, 0, 3.14159
3.14159, -3.14159, 1.19209e-07
-1.5708, 2.46192, 2.46192
-1.96559, 2.77592, 2.02578
-1.5708, 2.46192, 2.46192
4.71239, -2.46192, -2.46192
0, 0, 3.14159
3.14159, -3.14159, 1.19209e-07
-1.5708, 2.46192, 2.46192
-1.96559, 2.77592, 2.02578
-1.5708, 2.46192, 2.46192
4.71239, -2.46192, -2.46192
0, 0, 3.14159
3.14159, -3.14159, 1.19209e-07
-1.5708, 2.46192, 2.46192
-1.96559, 2.77592, 2.02578
-1.5708, 2.46192, 2.46192
4.71239, -2.46192, -2.46192
0, 0, 3.14159
3.14159, -3.14159, 1.19209e-07
-1.5708, 2.46192, 2.46192
-1.96559, 2.77592, 2.02578
-1.5708, 2.46192, 2.46192
4.71239, -2.46192, -2.46192
0, 0, 3.14159
3.14159, -3.14159, 1.19209e-07
-1.5708, 2.46192, 2.46192
-1.96559, 2.77592, 2.02578
-1.5708, 2.46192, 2.46192
4.71239, -2.46192, -2.46192
0, 0, 3.14159
3.14159, -3.14159, 1.19209e-07
-1.5708, 2.46192, 2.46192
-1.96559, 2.77592, 2.02578
-1.5708, 2.46192, 2.46192
4.71239, -2.46192, -2.46192
0, 0, 3.14159
3.14159, -3.14159, 1.19209e-07
-1.5708, 2.46192, 2.46192
-1.96559, 2.77592, 2.02578
-1.5708, 2.46192, 2.46192
4.71239, -2.46192, -2.46192
0, 0, 3.14159
3.14159, -3.14159, 1.19209e-07
-1.5708, 2.46192, 2.46192
-1.96559, 2.77592, 2.02578
-1.5708, 2.46192, 2.46192
4.71239, -2.46192, -2.46192
0, 0, 3.14159
```

[image]

xgerrmann, 2019-06-05:
For anyone encountering the same issue: It seems the joint was locked in the exact opposite direction of one of the joint axes due to a mixup of parent and child frame. It's fixed now.

closed 2019-06-05 by xgerrmann

lasagnaphil, 2019-12-12:
Although the issue is closed, I right now have a similar problem with erratic behavior of Twist, SwingX, and SwingY values. I've made sure that all of the parent and child frames have the same identity orientation when the articulated body is created, but this issue comes up when I try to apply joint rotations to the articulation via the PxArticulationCache. Am I doing something wrong here? It would be great if you could help.

xgerrmann, 2019-12-12:
@lasagnaphil I am not woeling on huis project anymore and its been too long to dive in again and help. So unfortunately I cannot help you here. Good luck!