NVIDIAGameWorks/PhysX #134: Need help for one day (paid)

xgerrmann, 2019-06-05:
Hello, 

I would like to have some help. I'm am stuck and need to get things moving again.

Let me briefly explain my situation:

I am adding and removing elements to the end of a reduced coordinate articulation. 
The articulation represents a cable (in an airborne wind energy system). The addition and removal of elements currently is not smooth enough.

I need someone who has a decent amount of experience with Nvidia Physx, especially with articulations.

If anyone is interested, please let me know. 
[image]

shaneasd, 2019-06-16:
Not interested in any paid work but since no one else seems interested I might be able to offer some advice at least. 

When you say the top of the cable is connected to the origin with a D6 joint, what are the specifics of the D6 joint? Which axes are constrained/driven and what is the nature of the constraint/drive for these axes?

I notice that all your links appear to be the same length except the link connecting the cable to the origin. Is this intentional? It looks to me like when you insert a new link it adds it too close to the previous one and so has to rapidly separate the two, adding energy to the system. In very approximate terms it looks as if you're conceptually jamming a compressed spring in between two parts and then letting it go which is going to energize the system.

It's not clear to me how much of the movement in this animation is from the cable behaving strangely and how much of it is from the wind you mentioned. Is this wind constant? Would be helpful to be able to see the wind force vector in the animation.

It looks to me like your cable should have rotational springs (and probably dampers) between links. The fact that the links can fold over so easily looks unrealistic. However I don't think fixing this would fix your problem (though it may mask it).

xgerrmann, 2019-06-17:
Hello @shaneasd, 
thank you for taking the time to provide your look at things. 
Let me address your questions and suggestions one-by-one.

> what are the specifics of the D6 joint? Which axes are constrained/driven and what is the nature of the constraint/drive for these axes?

I align the joint y-axis with the last cable element, this is the driven axis and is set as a free axis. Both the x and z-axes are locked. Rotation around all axes (twist, swing1 and swing2) are free.
I am not sure what you mean with "what is the nature of the constraint/drive axes" though, so if I've not answered this question please let me know.

> I notice that all your links appear to be the same length except the link connecting the cable to the origin. Is this intentional?

Indeed the distances between all PxArticulation links are identical. It might look like the first one (with varying length) is a link, but it is not. The first 'link' is the driven joint with variable distance. 
* The connections between the links and the origin are visualized by a line. 
* The articulation links are visualized by a set of axes. 
* The axes at the origin are not axes of a PxArticulation link. 
* The variable distance of the D6 joint is intentional.

> It looks to me like when you insert a new link it adds it too close to the previous one and so has to rapidly separate the two, adding energy to the system. In very approximate terms it looks as if [...]

I only introduce a new articulation link to the tether when the closest link has a distance of 2 times the distance as between all the other elements. This leaves 1  full element length between the newly created link and the D6 joint.

> It's not clear to me how much of the movement in this animation is from the cable behaving strangely and how much of it is from the wind you mentioned. Is this wind constant? Would be helpful to be able to see the wind force vector in the animation.

The wind is turned off, drag as well. The movement is caused by: 
1. The increase of drive distance of the D6 joint
2. Gravity
3. Faulty element introduction

> It looks to me like your cable should have rotational springs (and probably dampers) between links.

Good point, this will definitely reduce shockwave propagation. I still believe though, that element introduction should be possible without introducing any shocks.

I hope I've answered all your questions. Please let me know if you have any further suggestions/ remarks or questions. I am happy to answer them all.

shaneasd, 2019-06-17:
So I think one issue might be that you seem to be treating the links as though they are points but they should be bodies with geometry. I guess technically in a simulation like this you don't care about the geometry itself per se but it would provide a useful visualization of where the actual constraint poses (or joint position for reduced coordinate articulations) are which you do care about.

Which kind of articulation are you using (maximal or reduced coordinates)? Which version of physx?

If you were to remove gravity and the D6 joint from the system and add joints when you hit a key (or trigger from the debugger or whatever) does that look correct?

It might be useful to know the exact transform of the closest link to the origin, when you add a problematic link, and the transform of the link added (before ticking)

I notice around the 15s mark in your youtube clip that one of the connecting lines seems to split in half. Is this a rendering artifact? If not, what exactly do these lines represent?
[image]