NVIDIAGameWorks/PhysX #135: compile result confused aboud MTd(multithread debug) and MDd(multithread debug dll)

RPG3D, 2019-06-06:
[image]

some is MTd some id MDd

then link error
[image]

RPG3D, 2019-06-06:
[image]

AlesBorovicka, 2019-06-06:
Hi,
 looking at the output, it says that PhysX is compiled with MT/MTd. Which seems to be true, as the libraries of PhysX are in bin/win.x86_64.vc141.mt
But your application seems to compile with MD (Main.obj). So either change your app to use MT, or switch PhysX to compile MD:
set NV_USE_STATIC_WINCRT to False

RPG3D, 2019-06-06:
got it thanks

RPG3D, 2019-06-06:
resolved

closed 2019-06-06 by RPG3D