NVIDIAGameWorks/PhysX #136: PhysX GPU Performance

syslot, 2019-06-06:
Hi,

I'm trying to use PhysX as a backend to do Robotic Simulation. And I found it cost more time when I do a simulation with GPU compared with CPU. So I did a profile. 

I did 500 simulation steps and analysis of the profile. The most time-consuming part of a simulation step in GPU mode is Sync data, while the average duration of computing is very short. (e.g.: GpuDynamics.Solve : 213 us)

|                                                    | GPU   | CPU   |
|----------------------------------------------------|-------|-------|
| Ave Step (Basic.simulate, ms)                      | 8.148 | 2.249 |
| Ave Step Sync (GpuDynamics.DMABackBodies.Sync, ms) | 6.335 | -     |

Here is one simulation step stack
[image]

How to solve this problem?

Zogrim, 2019-06-08:
Afaik, that is expected, as data needs to be re-synced between CPU and GPU.
Thus, GPU simulation is effective when only large amount of active bodies is present in the scene. Otherwise, more time is spend and sync rather then simulation itself.

syslot, 2019-06-09:
> Afaik, that is expected, as data needs to be re-synced between CPU and GPU.
> Thus, GPU simulation is effective when only large amount of active bodies is present in the scene. Otherwise, more time is spend and sync rather then simulation itself.

Hi Zogrim,

The IO latency should exist, but why it's so large? 

I also test the SnippetHelloGRB and the result shows huge difference.

|                                                    | GPU   | 
|----------------------------------------------------|-------|
| Ave Step (Basic.simulate, ms)                      | 8.670 |
| Ave Step Sync (GpuDynamics.DMABackBodies.Sync, ms) | 0.45| 

So I want to know how to optimize this problem?

kstorey-nvidia, 2019-06-10:
The GpuDynamics.DMABackBodies.Sync will report times that are longer than just the DMA cost. All computation on the GPU is asynchronous and this is the point that synchronises the memory transfer back to the host, so this will include the cost of any kernel execution that hadn't already finished by the time the CPU reached this stage of the solver. 

The times reported by GpuDynamics.solve are the launch costs for the kernels - this is how much time we spent in the driver issuing commands to the GPU and performing any data marshalling on the CPU.

DMA costs generally should not be very large. We can move around 10-15MB per ms using PCIe 16x. If you don't have sufficient PCIe lanes and the connection drops to 8x or lower, DMA overhead could get quite high (using only 1x or 2x PCIe could be prohibitively slow, but your results with the snippet suggest this shouldn't be the case for you otherwise you would have seen the same issues there - it is DMAing back results for about 8-10k rigid bodies IIRC). 

Would you be able to give more details about what you are simulating and the configurations involved (number of robots, number of degrees of freedom, number of solver iterations, what you GPU vs CPU is etc.) so we can help root find the cause of this?

syslot, 2019-06-10:
Hi, kstorey.

I only use one robot with [laikago](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_data/laikago/laikago.urdf). The solver is pgs with 4. 

I review the urdf file and find out that the collision geometry models are described by complex obj files. Does this cause the large DMA sync latency? 

When I put more than 512 robots in one environment, DMA Sync duration begins to rise. If one robot causes such a large delay, then why more robots (10, 100, 300) still keeps the same duration of DMA sync?

Finally, I also test another [robot](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_data/quadruped/quadruped.urdf) with simple collision geometry models on GPU with pgs solver,  the exception 'allocAligned out of sharedMemory' occurred again. 

I can provide the source code if it helps.

kstorey-nvidia, 2019-06-11:
The numbers involved seem very high on both CPU and GPU given the complexity of the robot and the number of solver iterations being used. Are you definitely using either a checked/release/profile build of PhysX? The debug build is useful for debugging but the debug GPU module is ~100x slower than its release/profile/checked counterparts.

It is normal/expected that GPU is slower than CPU simulating small problems. CPUs and GPUs are optimized for different problems: CPUs are optimized to minimize latency with a small number of very fast cores. A GPU has 1000s or simpler cores running at a lower frequency. It could be an order of magnitude slower at computing a single thing, but it pulls ahead with larger problem domains because it can solve problems with 10s of 1000s of threads simultaneously (if the problems are large enough to tip the balance in the favour of the GPU).

The analogy is like trying to transport passengers from one island to another, having the option of a 4-person speedboat or a 10,000 person ferry. The ferry takes 10x longer to travel between islands than the speedboat. If you want to move just a few people, the speedboat is easily going to get everyone there faster. Once the number of passengers reaches a critical point, the ferry is the faster option.

(For marketing purposes, please visualise the ferry in this case as a beautiful superyacht and not a rusted up heap of junk ;) )

syslot, 2019-06-20:
> The numbers involved seem very high on both CPU and GPU given the complexity of the robot and the number of solver iterations being used. Are you definitely using either a checked/release/profile [...]

Thanks for your reply and I will test more. Close it.

closed 2019-06-20 by syslot

matangpanchal, 2023-03-25:
> Hi,
>
> I'm trying to use PhysX as a backend to do Robotic Simulation. And I found it cost more time when I do a simulation with GPU compared with CPU. So I did a profile.
>
> I did 500 simulation steps [...]

Hi, 
I am facing similar problem, So for profileing I want to know how you did this profiling ? can you provide, tool name and may be few instructions?