NVIDIAGameWorks/PhysX #139: PxArticulationJointReducedCoordinate - Set Drive Target & Velocity

Tadinu, 2019-06-17:
https://gameworksdocs.nvidia.com/PhysX/4.0/documentation/PhysXGuide/Manual/Articulations.html#joint-positions-and-velocities

Can you give a practical example of how to use `PxArticulationJointReducedCoordinate`'s `setDriveTarget` & `setDriveVelocity` either independtly and/or together?

How do I order a joint to reach a position with some specific velocity?
I guess the stiffness and damping would be the elements contributing to the force generated as using 'setDriveTarget', and thus the acceleration, so how do I incorporate the use of `setDriveVelocity` in this case?

Thanks,

Tad.

kstorey-nvidia, 2019-06-17:
The stiffness term in the drive drives towards the drive target (position/angle). It introduces forces that are proportional to the joint angle error (drive target - joint position) to drive the joint towards zero error (joint angle == drive target).

The damping term drives the joint towards a target velocity. It applies forces proportional to the velocity error (drive velocity - joint velocity) to drive the joint towards zero joint velocity error (joint velocity == drive velocity).

Generally, if you are using a positional drive target, you need to have some stiffness term that is strong-enough to move the system to the target angle. If the drive is very stiff, there is a chance of overshooting and then oscillating around the target angle like a spring. The PD controller actually uses a spring formula that is implicitly damping, so it generally works well. However, you can optionally add some damping to reduce overshoot/oscillation. In this scenario, I would expect that the drive velocity would be zero.

If you are driving towards a target velocity, you will usually have only a damping term and no stiffness term, because in this case the drive velocity and drive target would very likely fight against each other in an undesirable way.

It may be that you can make these 2 terms play nicely together if you are setting a non-zero target velocity and target position that both lead to the same target angle (for example) but, in that case, there doesn't seem to be much point in setting the target velocity IMO because you should get the same results using a target pose and using either no damping, or a small amount of damping with zero target velocity.

Tadinu, 2019-06-17:
Thanks a lot for your clarification, @kstorey-nvidia! I also observed that with my stiffness and a bit damping setting, the joint could reach its target position quite well using just `setTargetDrive` only.
So I should stick with it then.

Tadinu, 2019-06-18:
@kstorey-nvidia, just an extra question, so it seems if I don't use `setDriveVelocity` at all, if I would speed up or slow down the joint in a  trajectory, the only way is to adjust `stiffness` and `damping` on the way?

kstorey-nvidia, 2019-06-18:
Right. If you want to speed up how responsive your drive is, you have to adjust the stiffness and make it stiffer. The force applied is proportional to error  error*stiffness - velocity * damping.

SetDriveVelocity is more for the case where you want to set a constant drive velocity and have the drive translate/spin for an amount of time. You could get the same effect using drive targets but that means that the application has to work out a target pose each frame.

Tadinu, 2019-06-19:
I'm clear now. Thank you!

closed 2019-06-19 by Tadinu