NVIDIAGameWorks/PhysX #140: Custom friction for dynamic rigidbodies

krnkn, 2019-06-18:
Hi, I would like to change velocity of a rigidbody after simulation is complete so that i can apply friction for that simulation step. I do not have contact at that point and i dont want to use colliders. Is it possible to add contact without colliders before simulation and apply friction onContactModify? Or should i follow another approach? I have tried adding friction force after simulation; however position of the rigidbody is already updated and i can only change veloctiy for the next simulation step. Also I tried to change linear and angular velocities of mBodyCore.mCore in OnAdvance callback with a little bit tricks, bu as it seems those velocities are not used to update pose, it is already determined at that point.

kstorey-nvidia, 2019-06-19:
We don't currently have an API that makes it very easy for you to do this. We are discussing internally and may consider adding something in the future.

However, if you use the split simulation approach - instead of calling simulate/fetchResults, you call collide/fetchCollision then advance/fetchResults - you are allowed to modify velocities or apply forces up until you call advance and have these influence the rigid bodies that frame.

You cannot get contact reports until during fetchResults() but you can get contact modify callbacks during the collide() stage, so you should be able to achieve what you have set out to do using this technique (although this is not an officially-supported feature yet so it may be a little cumbersome to work with). Please note that it is not considered safe to apply forces on the bodies during the contact modify callback, so you should probably update your own buffer of forces and then apply them to the rigid bodies between fetchCollision() and the call to advance().

Hope this helps

krnkn, 2019-06-25:
Thank you kstorey. I have solved it.

closed 2019-06-25 by krnkn

krnkn, 2019-06-25:
Just one more question to clarify: When i check velocity and pose after just one simulation step for a free falling body(starting with linear velocity zero), it seems pose is updated by using after simulation velocity like: next_position = old_position + next_velocity*dt. Is there a reason why it is not considered constant acceleration but constant velocity? I mean why not (old_velocity+next_velocity)*0.5f*dt? Is it because of  bodies that need to stop because of the collision and we do not want them to interpenetrate?

kstorey-nvidia, 2019-06-25:
PhysX uses semi-implicit Euler integration, which basically equates to v = u + a*dt. This new velocity is computed prior to the constraint solver being run and serves as an input for the solver. The solver computes a change to these velocities that satisfy any constraints (joints, contacts, restitution etc.). Once we have the final velocity from the solver, we integrate using that velocity using s = v*dt. The intention of this is that we integrate using velocities that satisfy constraints on the system.

In the case of a free-falling unconstrained rigid body, these results may differ slightly from an analytical solution, which would integrate using s=u*dt+0.5*a*dt^2. However, as dt becomes small, Euler integration converges towards the analytical solution.

The alternative of trying to solve both for velocity and position independently, which will complicate the problem significantly for relatively little gain given that time-steps are expected to be relatively small.