NVIDIAGameWorks/PhysX #144: Articulation Link and Actor(not Articulation) Connection

k-sonobe, 2019-06-20:
I'm trying to connect articulation links and actors (PxRigidStatic or PxRigidDynamic) with general joints(PxJoint). Here is a simple code excerpt: 

```
PxRigidStatic* hook = PxCreateStatic(PxGetPhysics(), transform, *shape);

PxArticulationLink* link = articulation->createLink(parent, pose);
PxRigidActorExt::createExclusiveShape(*link, geometry, *material);
PxRigidBodyExt::updateMassAndInertia(*link, density);

PxRevoluteJoint* rev_joint = PxRevoluteJointCreate(PxGetPhysics(), hook, hook_joint_pose, link, link_joint_pose);
```

However, sometimes the joint are not working. If you hang the "link" on the "hook", the "link" will fall off the "hook" and drop. Apparently, this happens when the dimension of the articulation link is small. Why does this happen?

kstorey-nvidia, 2019-06-20:
Can you provide the full settings that reproduce this please?

There are some limits in the articulations that attempt to keep things numerically stable, but those should only cause artefacts at the other end of the spectrum AFAIK - joints may not work with extremely heavy links because the response vectors become sufficiently small that we consider them degenerate. 

If you can provide a repro, we can root cause it and fix it

k-sonobe, 2019-06-21:
Thank you for the very fast response and helpful information.

Here is a more detailed code:

```
PxMaterial* material = physics->createMaterial(0.7f, 0.7f, 0.6f);

PxVec3 extents = PxVec3(0.03f, 0.03f, 0.03f);
PxBoxGeometry geom;
geom.halfExtents = extents;

/* create hook(Static Actor) */
PxRigidStatic* hook = PxCreateStatic(*physics, PxTransform(PxVec3(0.0f, 0.0f, 0.0f)), geom, *material);
scene->addActor(*hook);

/* create link(Articulation) */
PxReal link_mass = 1.0f;
PxReal link_density = link_mass / ((extents.x * 2.0f) * (extents.y * 2.0f) * (extents.z * 2.0f));

articulation = physics->createArticulationReducedCoordinate();
PxArticulationLink* link = articulation->createLink(nullptr, PxTransform(PxVec3(0.0f, -(extents.y * 2.0f), 0.0f)));
PxRigidActorExt::createExclusiveShape(*link, geom, *material);
PxRigidBodyExt::updateMassAndInertia(*link, link_density);
articulation->setSolverIterationCounts(4, 1);
scene->addArticulation(*articulation);

/* create Revolute Joint between hook and link */
PxRevoluteJoint* rev_joint = PxRevoluteJointCreate(*physics,
	hook, PxTransform(PxVec3(-extents.x, -extents.y, 0.0f)),
	link, PxTransform(PxVec3(-extents.x, extents.y, 0.0f))
);

```

This is a static actor and an articulation composed of a single link, and the corners of these actors are connected by revolute joint.

zemlifr, 2020-10-05:
Hi, is there any solution to this? I have similar problem as @k-sonobe only I use fixed joint to connect RigidDynamic and Articulation. 
When articulation is heavy, joint doesn't work. Root link is fixed on place, but whole articulation rotates.

I guess I am encounterig these artifacts @kstorey-nvidia mentioned?