NVIDIAGameWorks/PhysX #145: Gpu Articulation abnormal

syslot, 2019-06-20:
Hi,

I use Articulation with server links and joints (less than 20) to simulate one robot. When I run with CPU, it's all correct.

When I test on GPU, 

- One exception (allocAligned out of sharedMemory) described in https://github.com/NVIDIAGameWorks/PhysX/issues/129#issue-449840826 throws. 

- Another problem. When I change another Urdf and still use GPU, when finishing PxScene::simulate and PxScene::fetchResults (already set fetchResults with block = true and errorState return 0), I get all joint info from Articulation cache with **0**. 

How to solve such two problems?

kstorey-nvidia, 2019-06-21:
The first issue will require an update to the release. It's a bug in the GPU module that we have fixed internally and are working towards from a branch to trunk along with other improvements. We hope to have a bugfix release including this change very soon.

I will experiment with the other issue to see if I can reproduce. If I can, we will include a fix for that in that release also, otherwise I'll come back asking for a repro from you

syslot, 2019-06-21:
> The first issue will require an update to the release. It's a bug in the GPU module that we have fixed internally and are working towards moving back from a branch to trunk along with other improvements. We hope to have a bugfix release including this change very soon.
> 
> I will experiment with the other issue to see if I can reproduce. If I can, we will include a fix for that in that release also, otherwise I'll come back asking for a repro from you

Thanks, kstorey. 

Your support is professional, timely and amazing. Tell me what I can do if you need !

syslot, 2019-06-21:
Hi, kstorey.

Today I also find another exception. When I test on GPU with TGS solver, one crash occurred. Here is the core dump stack.

```
#0  0x00005650c29646a4 in void physx::Bp::boxPruningKernel<0>(physx::Bp::(anonymous namespace)::MBP_PairManager*, bool const*, unsigned int, unsigned int const*, physx::Bp::AABB_Xi const*, physx::Bp::AABB_YZn const*, unsigned int, unsigned int const*, physx::Bp::AABB_Xi const*, physx::Bp::AABB_YZn const*, physx::Bp::FilterGroup::Enum const*) ()
#1  0x00005650c296cd58 in physx::Bp::PersistentAggregateAggregatePair::findOverlaps(physx::Bp::(anonymous namespace)::MBP_PairManager&, physx::PxBounds3 const*, float const*, physx::Bp::FilterGroup::Enum const*, bool const*) ()
#2  0x00005650c29676fd in physx::Bp::AABBManager::updatePairs(physx::Bp::PersistentPairs&, physx::Bp::BpCacheData*) ()
#3  0x00005650c2967cb2 in physx::Bp::PersistentAggregateAggregatePair::update(physx::Bp::AABBManager&, physx::Bp::BpCacheData*) ()
#4  0x00005650c296e1c0 in physx::Bp::ProcessAggPairsParallelTask::runInternal() ()
#5  0x00005650c253df01 in physx::Cm::Task::run() ()
#6  0x00005650c255f241 in physx::Ext::CpuWorkerThread::execute() ()
#7  0x00005650c29b402a in physx::shdfnd::(anonymous namespace)::PxThreadStart(void*) ()
#8  0x00007f183d97e6db in start_thread (arg=0x7f1838f59700) at pthread_create.c:463
#9  0x00007f183afd588f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

syslot, 2019-07-16:
> The first issue will require an update to the release. It's a bug in the GPU module that we have fixed internally and are working towards moving back from a branch to trunk along with other [...]

Hi, kstorey.

Does this problem have some solutions now?

syslot, 2019-07-17:
Hi,  all.

Now I found that all of GPU problems I reported before ([1](https://github.com/NVIDIAGameWorks/PhysX/issues/129#issue-449840826), [2](https://github.com/NVIDIAGameWorks/PhysX/issues/136), [3](https://github.com/NVIDIAGameWorks/PhysX/issues/145)) are caused by GPU Dynamics. 

So now, I only use GPU to calculate BroadPhase and it works normally.

Close now and wait for next version of GPU Dynamics function :)

kstorey-nvidia, 2019-07-17:
Thanks

We are working on an updated release hopefully next week that will include fixes for these issues.

Kier

syslot, 2019-08-05:
> Thanks
> 
> We are working on an updated release hopefully next week that will include fixes for these issues.
> 
> Kier

Hi, Kier,

Is it possible to release a bug fixed version PhysX this week?
I'm still waiting to test my framework, and GPU acceleration is the most important part.

:)

syslot, 2019-08-18:
> Thanks
> 
> We are working on an updated release hopefully next week that will include fixes for these issues.
> 
> Kier

Hi Kier, 

I am very happy that PhysX has updated the new version, but it seems that I still can't use GPU_dynamics with articulation. Is there a problem with this?