NVIDIAGameWorks/PhysX #149: Is there a way to apply forces directly to articulation joints?

arpspoof, 2019-07-02:
Hi,

I'm currently working on an articulation connected by joints. According to the documents I've looked into, seems like the only way to drive the joint is by specifying target positions so that the default PD drive will apply the corresponding force. However, I'm wondering is there a way to apply my own calculated forces directly to the joints without going through the built-in PD drive? Or am I missing anything?

Thanks for any help!

kstorey-nvidia, 2019-07-03:
Please take a look at PxArticulationCache. It has a buffer for you to set joint forces on each joint degree of freedom. This is also how you gain access to the joint velocities, joint positions, joint accelerations etc. All these operate in reduced/generalised coordinates.

arpspoof, 2019-07-03:
Thank you for the fast response and the helpful information. I'll take a look at that!

eyepatch61, 2019-07-04:
thank you for the response. If I am able to get all points. I want to write it in a file. somehow file handling is not working in it. anyone knows about this issue? I hav e included all headers necessary. I am trying to use file handling in snippetMain function. Is that right?

closed 2019-07-13 by arpspoof