NVIDIAGameWorks/PhysX #150: PxVehicleSuspensionSweeps generate a wrong vertical force when the vehicule is rolling on 2 wheels.

thomaspierson, 2019-07-03:
Hello,

I’m trying to implement the full wheels contact mod instead of raycasts using  PxVehicleSuspensionSweeps().
I got an sort of bad side effect: a wrong vertical force is applied when the vehicle start to roll (on 2 wheels). This make it unable to fall on the side in a realistic way.

Someone else post this video which describe exactly this issue:

- https://www.youtube.com/watch?v=iCGNdJCY18Y

It’s like in the suspension force is applied in the wrong direction related to the ground and not related to the vehicle body or somthing like that.

I’m using PhysX 4.1.

Thanks in advance for your help.

gyeomannvidia, 2019-08-19:
Can you try to repro this with raycasts instead of sweeps?

Cheers,

Gordon

DaveGravel, 2019-09-02:
Other problem with PxVehicleSuspensionSweeps(abatchQr, 1, vehicles, sweepResultsSize, sweepResults, 16, NULL, 1.0f, 1.01f);
Exemple if I place a dynamic object on the vehicle chassis over a wheel position, The Sweep ray seen to interfering with the dynamic object. And the wheel start to do penetration in the floor.
I can avoid this by changing the 1.0f,1.01f values with something like 0.5f, 0.51f, But again it cause problem the wheel become a lot more bumpy when it touch a edge corner.

With the PxVehicleSuspensionRaycasts(abatchQr, 1, vehicles, raycastResultsSize, raycastResults); 
it working pretty good, But with this mode the wheel is more bumpy with the edge corner.

The SuspensionSweeps with values 1.0f, 1.01f is working better with the edge and when you like to roll over dynamics object like little box. But something go wrong when a dynamics object go on the vehicle over a wheel position.

If someone need a video about this problem I can make one.
I use 4.1 too.

DaveGravel, 2019-09-02:
Ok here a video for show the problem with Sweeps mode:
https://www.youtube.com/watch?v=8K5uMyfzeHM

PxVehicleSuspensionSweeps with values 1.0f, 1.01f:
With this mode the two wheels test seen to work good for me.
I can roll over dynamic objects and it don't seen to cause heavy bumpy effet when the wheel hit the edges.
The only problem I can't place object on the chassis vehicle because the ray start to interfering and do penetration in the floor.
I can use object with any mass and it do the same thing.

PxVehicleSuspensionSweeps with values 0.5f, 0.51f:
With this mode the two wheels test seen to work good too.
When I roll over dynamic objects the wheels get a more heavy bumpy effect, 
But with this values I can place object over the chassis and it seen to avoid the interfering problem and don't do penetration in the floor.

PxVehicleSuspensionRaycasts
With this mode the two wheels test seen to work good too. 
When I roll over dynamic object the wheels do a more heavy bumpy effect and the collision look a bit more bad because this mode do only one ray under the wheels. 
With this mode I can place dynamic object over the chassis without any problem.

In the video at around 10:00min I test the two wheels mode with PxVehicleSuspensionRaycasts and it seen to work ok for me if i'm not wrong.

At around 10:50min I do the same two wheels test with the PxVehicleSuspensionSweeps  and again it seen to work ok for me.

Sorry about my english.

gyeomannvidia, 2019-09-17:
You will need to use the query filtering to filter out hits against unwanted objects.   I'd recommend rigid body collision with dynamic objects and suspension response to static objects.

DaveGravel, 2019-09-18:
Thanks for the reply.
Hum, I already use filter maybe I do something wrong in my setup.
I can't use filter for totally avoid the dyn objects by the SuspensionSweeps or maybe I don't understand right the way how it work, The main goal is to have the tires rolling over dynamics objects. 
The tires collision behaving a lot better when I use SuspensionSweeps.
With the normal raycast suspension tires the result don't look so good and it is normal because the ray have only one contact point at time.
I can retry with the rigid body collision but in my last test with rigid body the wheel is a lot bumpy over the floor.
The bumpy effect is less present with the sphere shape, But the wheel simultion from sphere is really bad idea.
With cylinder or similar shape the wheel is a lot bumpy.
Thanks for the suggestions, I can do more tests with it.

thomaspierson, 2019-12-18:
> Can you try to repro this with raycasts instead of sweeps?

@gyeomannvidia Thank you for your response. I just tested again and no with raycasts unstead of sweeps it’s ok, no extra force generated.

@DaveGravel It seems like a different issue, please open a different issue if you need.