NVIDIAGameWorks/PhysX #151: Jetson Nano

alanjfs, 2019-07-08:
Hi @PhysX,

I'm working on a robotics project involving PhysX and was wondering whether it's possible/supported running PhysX on the [Jetson Nano](https://developer.nvidia.com/embedded/jetson-nano-developer-kit) GPU? It's running Debian on a 64-bit ARM processor, which I figure isn't too dissimmilar to the support for Android or iOS (disclaimer: my experience building for ARM is limited). The [prerequisites for Android](https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/platformreadme/android/readme_android.html) seem heavily Android-specific, which gives me pause from trying this out on the device (not to mention buying it in the first place).

AlesBorovicka, 2019-07-15:
Hi,
 We dont support the GPU on the Jetson Nano, though you should be able to run the SDK there on the ARM CPU. This is supported and you can generate the required projects by running:
./generate_projects.sh linux-aarch64 
Note that you will need the aarch64 gcc compiler installed on your host linux machine. 
Regards,
Ales

alanjfs, 2019-07-22:
Thanks @AlesBorovicka, closing this for now as it was the GPU on the Nano that was of interest rather than the CPU.

closed 2019-07-22 by alanjfs