NVIDIAGameWorks/PhysX #152: Visual Studio 2019 project generator missing

PlasmaMichael, 2019-07-15:
Hi,
"generate_projects.bat" can not generate a project for Visual Studio 2019.
All the best.

AlesBorovicka, 2019-07-16:
Hi,
 that is correct. We are still internally testing the Visual Studio 2019, there are still opened issues. If resolved, the next update should have the support included.
Regards,
Ales

PlasmaMichael, 2019-07-16:
For now I simply modified the python script to unblock me and it worked fine for x64 but not x86, which is fine for me since I focus on x64, x86 is not anymore even supported by FBX SDK.

fmehmetun, 2019-08-05:
> Hi,
> that is correct. We are still internally testing the Visual Studio 2019, there are still opened issues. If resolved, the next update should have the support included.
> Regards,
> Ales

Is there any update for this?

AlesBorovicka, 2019-08-05:
Yes, in next update the support will be added. The update should happen this or next week.