NVIDIAGameWorks/PhysX #153: No suffix on the debug libs

PlasmaMichael, 2019-07-15:
Hi,
There is no suffix for the debug libs using Visual Studio compiler and I guess this is the same for the other compilers.
All the best.

AlesBorovicka, 2019-07-16:
Hi,
 yes, this was a design decision, there are no suffixes, you should be able to toggle between different configuration libraries without the need to change the library names. The configuration suffix moved a directory name. 
Regards,
Ales

PlasmaMichael, 2019-07-16:
Ah yes... the problem is my binary folder contains all the DLL and a suffix would have been good like before, do you have a recommendation to have the toggle easy on this case? Unreal Engine for example also has one binary folder, it's not split in two (debug and release).

AlesBorovicka, 2019-07-16:
You can try to modify the preset that you use (e.g. vc14win64.xml) 
Add these switches:
    <cmakeSwitch name="NV_APPEND_CONFIG_NAME" value="True" comment="Append config " />
    <cmakeSwitch name="NV_USE_GAMEWORKS_OUTPUT_DIRS" value="False" comment="Use new GameWorks folder structure for binary output." />
    <cmakeSwitch name="PX_COPY_EXTERNAL_DLL" value="False" comment="Copy external dlls into SDK bin directory" />

Nanoatic, 2020-02-12:
`<cmakeSwitch name="NV_APPEND_CONFIG_NAME" value="True" comment="Append config " />` this is sufficient to solve my issue with renaming.