NVIDIAGameWorks/PhysX #154: GPU Function Register Error

syslot, 2019-07-17:
Hi, 

I try to use PhysX with GPU as backend to test the Reinforcement Learning environment and based on [erwans' work](https://github.com/erwincoumans/bullet3/tree/physx4), I implement an OpenAI Gym with PhysX backend in Python. 

When I import PhysX and GPU Version [PyTorch](https://github.com/pytorch/pytorch) or [Tensorflow](https://github.com/tensorflow/tensorflow), PhysX's cuda_manager seems work abnormally. When I switch to CPU Version PyTorch or Tensorflow, PhysX works well.  

Can anyone know is this an implement error? Thanks.

syslot, 2019-07-17:
Here is the some message reported by PhysX.

```
Could not find registered CUDA function 'updateBodyExternalVelocitiesLaunch'. in file:/buildAgent/work/9b8aa436d5e8ada1/source/cudamanager/src/CudaKernelWrangler.cpp line:98
GPU translateAABBs fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:785
GPU compute end point kernel fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:930
GPU compute end point between block kernel fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:934
GPU updateCreatedPairsKernel fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:898
GPU radix sort fail to launch kernel!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:630
GPU radix sort fail to launch kernel!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:635
GPU radix sort fail to launch kernel!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:630
GPU radix sort fail to launch kernel!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:635
GPU radix sort fail to launch kernel!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:630
GPU radix sort fail to launch kernel!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:635
GPU radix sort fail to launch kernel!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:630
GPU radix sort fail to launch kernel!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:635
GPU radix sort fail to launch kernel!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:630
GPU radix sort fail to launch kernel!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:635
GPU radix sort fail to launch kernel!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:630
GPU radix sort fail to launch kernel!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:635
GPU radix sort fail to launch kernel!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:630
GPU radix sort fail to launch kernel!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:635
GPU radix sort fail to launch kernel!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:630
GPU radix sort fail to launch kernel!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:635
GPU compute end point kernel fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:930
GPU compute end point between block kernel fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:934
GPU initialize SapBox Kernel fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:759
GPU create region fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:974
GPU compute start and active end point kerenl fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:982
GPU output active histogram fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:987
GPU compute region histogram kerenl fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:997
GPU output region histogram fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:1001
GPU write out start and active end point kerenl fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:1034
GPU compute start and active end point kerenl fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:1038
GPU output active histogram fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:1042
GPU write out handle/region overlap checkes kernel fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:1110
GPU compute overlap checks function kernel fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:1120
GPU output overlap checks function kernel fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:1130
GPU found pair kernel fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:1141
GPU clearNewFlag kernel fail to launch!!
 in file:/buildAgent/work/9b8aa436d5e8ada1/source/gpubroadphase/src/PxgCudaBroadPhaseSap.cpp line:1186
```

phil-hawkins, 2020-10-29:
I'm seeing the same behavior.  I'm using PhysX 4.1 and cudatoolkit 10.1.

@syslot, did you have any success getting this to work?