NVIDIAGameWorks/PhysX #155: Android NDK Project generation fails

elgunhuseynli, 2019-07-19:
I installed latest Android NDK r20 to my Windows 10 PC and tried to generate projects file but it failed with this error. It seems like it cant build (or generate) project with latest NDK version.
```
Using preset xml: buildtools/presets/public/android.xml
Target platform: android using compiler: clang
PM_CMakeModules_PATH: C:/PhysX/physx/../externals/CMakeModules
PM_PATHS: C:/PhysX/physx/../externals/CMakeModules;C:/PhysX/physx/../externals/targa
Cmake: cmake.exe
Not searching for unused variables given on the command line.
CMake Warning at C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:323 (message):
  Could not determine machine name for compiler from
  C:/Users/elgun/AppData/Local/Android/Sdk/ndk-bundle/toolchains/aarch64-linux-android-4.9/prebuilt/windows-x86_64
Call Stack (most recent call first):
  C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:550 (__DETECT_TOOLCHAIN_MACHINE_NAME)
  C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:607 (__GLOB_NDK_TOOLCHAINS)
  C:/Program Files/CMake/share/cmake-3.15/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:34 (project)

CMake Warning at C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:323 (message):
  Could not determine machine name for compiler from
  C:/Users/elgun/AppData/Local/Android/Sdk/ndk-bundle/toolchains/arm-linux-androideabi-4.9/prebuilt/windows-x86_64
Call Stack (most recent call first):
  C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:550 (__DETECT_TOOLCHAIN_MACHINE_NAME)
  C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:607 (__GLOB_NDK_TOOLCHAINS)
  C:/Program Files/CMake/share/cmake-3.15/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:34 (project)

CMake Warning at C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:323 (message):
  Could not determine machine name for compiler from
  C:/Users/elgun/AppData/Local/Android/Sdk/ndk-bundle/toolchains/x86-4.9/prebuilt/windows-x86_64
Call Stack (most recent call first):
  C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:550 (__DETECT_TOOLCHAIN_MACHINE_NAME)
  C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:607 (__GLOB_NDK_TOOLCHAINS)
  C:/Program Files/CMake/share/cmake-3.15/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:34 (project)

CMake Warning at C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:323 (message):
  Could not determine machine name for compiler from
  C:/Users/elgun/AppData/Local/Android/Sdk/ndk-bundle/toolchains/x86_64-4.9/prebuilt/windows-x86_64
Call Stack (most recent call first):
  C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:550 (__DETECT_TOOLCHAIN_MACHINE_NAME)
  C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:607 (__GLOB_NDK_TOOLCHAINS)
  C:/Program Files/CMake/share/cmake-3.15/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:34 (project)

CMake Error at C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:616 (message):
  Could not find any working toolchain in the NDK.  Probably your Android NDK
  is broken.
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.15/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:34 (project)

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
Not searching for unused variables given on the command line.
CMake Warning at C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:323 (message):
  Could not determine machine name for compiler from
  C:/Users/elgun/AppData/Local/Android/Sdk/ndk-bundle/toolchains/aarch64-linux-android-4.9/prebuilt/windows-x86_64
Call Stack (most recent call first):
  C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:550 (__DETECT_TOOLCHAIN_MACHINE_NAME)
  C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:607 (__GLOB_NDK_TOOLCHAINS)
  C:/Program Files/CMake/share/cmake-3.15/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:34 (project)

CMake Warning at C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:323 (message):
  Could not determine machine name for compiler from
  C:/Users/elgun/AppData/Local/Android/Sdk/ndk-bundle/toolchains/arm-linux-androideabi-4.9/prebuilt/windows-x86_64
Call Stack (most recent call first):
  C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:550 (__DETECT_TOOLCHAIN_MACHINE_NAME)
  C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:607 (__GLOB_NDK_TOOLCHAINS)
[... 101 lines cut ...]
  C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:550 (__DETECT_TOOLCHAIN_MACHINE_NAME)
  C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:607 (__GLOB_NDK_TOOLCHAINS)
  C:/Program Files/CMake/share/cmake-3.15/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:34 (project)

CMake Warning at C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:323 (message):
  Could not determine machine name for compiler from
  C:/Users/elgun/AppData/Local/Android/Sdk/ndk-bundle/toolchains/x86_64-4.9/prebuilt/windows-x86_64
Call Stack (most recent call first):
  C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:550 (__DETECT_TOOLCHAIN_MACHINE_NAME)
  C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:607 (__GLOB_NDK_TOOLCHAINS)
  C:/Program Files/CMake/share/cmake-3.15/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:34 (project)

CMake Error at C:/PhysX/externals/cmakemodules/android/android.toolchain.cmake:616 (message):
  Could not find any working toolchain in the NDK.  Probably your Android NDK
  is broken.
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.15/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:34 (project)

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
```

AlesBorovicka, 2019-07-19:
We will try to move to native cmake android support, but I am not sure when. It wont come in next release, we can try for the release after.

stevenhaggerty, 2020-01-15:
Any chance this can be bumped up the priority? 

Since the NDK moved from GCC to clang the toolchain detection fails as it looks for "gcc" in the toolchain binaries, therefore it can't generate the projects.

As it stands no one using the current android studio can build PhysX.