NVIDIAGameWorks/PhysX #156: Physx4.1 Crash with more than 65k rigid bodies and d6 constraints

iaiotom, 2019-07-23:
Hello, I have a scene with more than 3000 ragdolls with around 30 rbs and 25 d6 joint each. I have a constant crash in:
void DynamicsContext::setDescFromIndices(PxSolverConstraintDesc& desc, IG::EdgeIndex edgeIndex, const IG::SimpleIslandManager& islandManager,
	PxU32* bodyRemap, const PxU32 solverBodyOffset)
…
desc.bodyADataIndex = Ps::to16(index + 1);

Since there is a 16 bit conversion it raise an assert there when it pass the 65k index. Is there an internal limitation of 65k rbs in physx?

Thanks

kstorey-nvidia, 2019-07-23:
Thanks for reporting. We previously had a limitation of 64k but this limitation was removed in PhysX 4. It unfortunately looks like a remnant of this in the joint setup code slipped through the cracks. The contact constraint code does not convert to a u16.

Could you please remove the Ps::to16 from the lines:

desc.bodyADataIndex = Ps::to16(index + 1);

and 

desc.bodyBDataIndex = Ps::to16(index + 1);

Just replace both with "index + 1". I will verify on my side that this resolves the issue.

The fix will be included in our next release

iaiotom, 2019-07-23:
Thank you Keir, yes that fixed for the constraints but I found the same problems with the materials. 
bool NpMaterialManager::setMaterial(NpMaterial& mat)
…
const PxU16 materialIndex = Ps::to16(poolID);
…
It's not a big problem for us since I'm now hashing and reusing materials with the same parameters but it will be good to have this one fixed as well in the next release.
Thank  you
Alan

kstorey-nvidia, 2019-07-23:
Limiting material indices to 16-bit indices is more of a design decision than a bug. It's really unlikely that you would ever need that many unique materials and it is far more memory and cache-efficient to share materials. 

If we were to support > 65536 materials, we would need to expand the material index in the shapes to 32-bit. IIRC, this would end up spilling past a 16-byte boundary and make each shape occupy an additional 16 bytes of memory.

iaiotom, 2019-08-07:
I understand, thanks for the explanation.

closed 2019-08-07 by iaiotom