NVIDIAGameWorks/PhysX #157: generate_projects.bat errors

AuroraHalsey, 2019-07-26:
Hello,

I'm not really sure on how to go about creating a PhysX project from scratch. I've been following the [guide](https://gameworksdocs.nvidia.com/PhysX/4.0/documentation/PhysXGuide/Manual/BuildingWithPhysX.html), but have stumbled on the first step.

I've been trying to generate a PhysX project for Visual Studio 2017. I have Python and CMake installed, along with all the Visual Studio packages, and yet I get a bundle of errors each time. I'm not able to interpret and fix it myself, so help would be appreciated.

Thanks

```
I:\Github Repositories\PhysX\physx>generate_projects.bat

I:\Github Repositories\PhysX\physx>exit /b 0
Preset parameter required, available presets:
(0) android <--- Android-19, armeabi-v7a with NEON PhysX SDK
(1) vc12win32 <--- VC12 Win32 PhysX general settings
(2) vc12win64 <--- VC12 Win64 PhysX general settings
(3) vc14win32 <--- VC14 Win32 PhysX general settings
(4) vc14win64 <--- VC14 Win64 PhysX general settings
(5) vc15uwp32 <--- VC15 UWP 32bit PhysX general settings
(6) vc15uwp64 <--- VC15 UWP 64bit PhysX general settings
(7) vc15uwparm32 <--- VC15 UWP 32bit PhysX general settings
(8) vc15uwparm64 <--- VC15 UWP ARM 64bit PhysX general settings
(9) vc15win32 <--- VC15 Win32 PhysX general settings
(10) vc15win64 <--- VC15 Win64 PhysX general settings
Enter preset number: 10
Running generate_projects.bat vc15win64

I:\Github Repositories\PhysX\physx>exit /b 0
Using preset xml: buildtools/presets/public/vc15win64.xml
Target platform: win64 using compiler: vc15
PM_CMakeModules_PATH: I:/Github Repositories/PhysX/physx/../externals/CMakeModules
PM_PATHS: I:/Github Repositories/PhysX/physx/../externals/CMakeModules;I:/Github Repositories/PhysX/physx/../externals/targa
Cmake: cmake.exe
Not searching for unused variables given on the command line.
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.17134.
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:34 (project):
  No CMAKE_C_COMPILER could be found.

CMake Error at CMakeLists.txt:34 (project):
  No CMAKE_CXX_COMPILER could be found.

-- Configuring incomplete, errors occurred!
See also "I:/Github Repositories/PhysX/physx/compiler/vc15win64/CMakeFiles/CMakeOutput.log".
See also "I:/Github Repositories/PhysX/physx/compiler/vc15win64/CMakeFiles/CMakeError.log".
```
Log Files:
[CMakeError.log](https://github.com/NVIDIAGameWorks/PhysX/files/3436108/CMakeError.log)
[CMakeOutput.log](https://github.com/NVIDIAGameWorks/PhysX/files/3436109/CMakeOutput.log)

AlesBorovicka, 2019-07-29:
Hi,
 the log says that you dont have VS2017 compiler:
`-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:34 (project):
  No CMAKE_C_COMPILER could be found.`

Furthermore in the CMakeError.log there is the output of the test compilation that cmake does. The error is:
LINK : fatal error LNK1104: cannot open file 'MSVCRTD.lib'
I think its related to this warning:
I:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppBuild.targets(402,5): warning MSB8038: Spectre mitigation is enabled but Spectre mitigated libraries are not found.  Verify that the Visual Studio Workload includes the Spectre mitigated libraries.  See https://aka.ms/Ofhn4c for more information. [I:\Github Repositories\PhysX\physx\compiler\vc15win64\CMakeFiles\3.15.0\CompilerIdC\CompilerIdC.vcxproj]

Check this thread for some help:
https://stackoverflow.com/questions/6228112/link-fatal-error-lnk1104-cannot-open-file-msvcrtd-lib

Looks like you are missing some features from the VS install?
Regards,
Ales

AuroraHalsey, 2019-07-29:
Hello,

Thanks for the reply. I discovered that spectre mitigation was enabled without the correct libraries when trying to make VS2017 work with a sample project. I've since fixed the problem and everything is working, so far at least.

Thanks for the help.

Rora

closed 2019-07-29 by AuroraHalsey