NVIDIAGameWorks/PhysX #163: Visual Studio 2019 compiler bug

RT222, 2019-08-13:
Hello,

When I try to build PhysX 4.1.1 with Visual Studio 2019, I get the following errors :

```
Error	LNK1181	cannot open input file 'D:\Resources\Libraries\vcpkg\buildtrees\physx\src\53e20a06fa-2d853903fb\physx\compiler\vc16win32\sdk_source_bin\LowLevelDynamics.dir\profile\DyArticulationHelper.obj'	PhysX	D:\Resources\Libraries\vcpkg\buildtrees\physx\src\53e20a06fa-2d853903fb\physx\compiler\vc16win32\sdk_source_bin\LINK	1	
Error	C1001	 An internal error has occurred in the compiler.
(compiler file 'd:\agent\_work\3\s\src\vctools\Compiler\Utc\src\p2\main.c', line 160)
 To work around this problem, try simplifying or changing the program near the locations listed above.
Please choose the Technical Support command on the Visual C++
 Help menu, or open the Technical Support help file for more information
  CL!InvokeCompilerPass()+0xf1297
  CL!InvokeCompilerPass()+0xf1297
  CL!CloseTypeServerPDB()+0x7ae5b	LowLevelDynamics	D:\Resources\Libraries\vcpkg\buildtrees\physx\src\53e20a06fa-2d853903fb\physx\source\foundation\include\windows\PsWindowsInlineAoS.h	366	
Error	C1001	 An internal error has occurred in the compiler.
(compiler file 'd:\agent\_work\3\s\src\vctools\Compiler\Utc\src\p2\main.c', line 160)
 To work around this problem, try simplifying or changing the program near the locations listed above.
Please choose the Technical Support command on the Visual C++
 Help menu, or open the Technical Support help file for more information
  CL!InvokeCompilerPass()+0xf1297
  CL!InvokeCompilerPass()+0xf1297
  CL!CloseTypeServerPDB()+0x7ae5b
cl : Command line error D8040 : error creating or communicating with child process	LowLevelDynamics	D:\Resources\Libraries\vcpkg\buildtrees\physx\src\53e20a06fa-2d853903fb\physx\source\foundation\include\windows\PsWindowsInlineAoS.h	366	
```

It happens on release, profile and checked configurations (not debug), and only when building for x86 platforms (x64 is working fine).

It looks like MSVC have a problem with this function in PsWindowsInlineAoS.h

```
PX_FORCE_INLINE Vec3V V3LoadA(const PxF32* const f)
{
	ASSERT_ISALIGNED16(f);
	return V4ClearW(_mm_load_ps(f));
}
```

By using this small workaround, I've been able to make everything build with no problem : 

```
PX_FORCE_INLINE Vec3V V3LoadA(const PxF32* const f)
{
	ASSERT_ISALIGNED16(f);
	PxF32 msvc_1_4_22_bug_fix = *f;
	return V4ClearW(_mm_load_ps(&msvc_1_4_22_bug_fix));
}
```

It appears to be a compiler bug. I'm using Visual Studio 2019 with MSVC build tools 1.4.22. I wanted to report it to you first, but the next step is probably reporting it to Microsoft.

AlesBorovicka, 2019-08-14:
Not sure if its a problem of a newer VS version or older. Please, what is your Visual Studio 2019 version? That version you posted, I am still not sure to what VS version it links. I am using VS 16.1.6 and it seems to compile just fine.

RT222, 2019-08-14:
It seem to be a problem with a newer VS version then, because mine is 16.2.2.

AlesBorovicka, 2019-08-21:
Hmm today's update to 16.2.3 seems to not fix the issue. I am so far not able to isolate the issue to send a simplified repro case. 
Thanks for the workaround!