NVIDIAGameWorks/PhysX #164: PsAllocator.h(40,11) : fatal error C1083: Cannot open include file: 'typeinfo.h': No such file or directory

ericwj, 2019-08-16:
It'll build with [`#include <typeinfo.h>`](https://github.com/NVIDIAGameWorks/PhysX/blob/4050bbfdc2699dfab7edbf0393df8ff96bbe06c5/physx/source/foundation/include/PsAllocator.h#L38-L44) changed to `#include <typeinfo>` on line 40.

StephanTLavavej, 2019-10-09:
This is because I removed the non-Standard header `<typeinfo.h>` from MSVC. Including `<typeinfo>` is the correct fix, and is compatible with all versions of MSVC (going back at least a decade).

SeanMaxhell, 2019-11-05:
I have a similar Problem with Visual Studio 2019 and Unreal Engine Oculus Integration.
But I don't understand what I have to do to solve it.
Any Help?

ericwj, 2019-11-05:
If you have 'a similar' problem then you have to be more specific or even open another issue.

If you have the exact same problem then you can literally open `PsAllocator.h`, go to line 40 and remove `.h` from that line.

kornellvarga, 2020-03-20:
> It'll build with [`#include <typeinfo.h>`](https://github.com/NVIDIAGameWorks/PhysX/blob/4050bbfdc2699dfab7edbf0393df8ff96bbe06c5/physx/source/foundation/include/PsAllocator.h#L38-L44) changed to `#include <typeinfo>` on line 40.

Hmm its all fine and works, but what I do not understand is when a change like this happens why is there no follow-up for references such as the PsAllocator.h.
It would cost 5 min to push a changed version, and save half an hour each person who is as unexperienced as I am.

xcplay, 2020-08-16:
https://blog.csdn.net/weixinyi21cn/article/details/108035285

ericwj, 2022-08-01:
Looks like #181 fixed this.

Haven't tested it so let me know if its true then I'll close this.

mastercoms, 2022-08-01:
Yep it's fixed

ericwj, 2022-08-01:
ty

closed 2022-08-01 by ericwj