NVIDIAGameWorks/PhysX #165: return const member as non-const

erwincoumans, 2019-08-16:
~~~~~~~~~~~~~~~
src/PhysX/physx/source/geomutils/src/gjk/GuGJKType.h:154:76: error: binding reference of type ‘physx::shdfnd::aos::PsMatTransformV&’ to ‘const physx::shdfnd::aos::PsMatTransformV’ discards qualifiers
   PX_FORCE_INLINE Ps::aos::PsMatTransformV& getRelativeTransform(){ return mAToB; }
~~~~~~~~~~~~~~~

trivial to fix:
~~~~~~~~~~~~~~~
-               PX_FORCE_INLINE Ps::aos::PsMatTransformV& getRelativeTransform(){ return mAToB; }
+               PX_FORCE_INLINE const Ps::aos::PsMatTransformV& getRelativeTransform() const { return mAToB; } 
~~~~~~~~~~~~~~~

Pierre-Terdiman, 2019-08-27:
Thanks! Submitted to our trunk, will appear in next GitHub release.

erwincoumans, 2020-01-17:
Thanks Pierre, but the fix is not in this 4.1.1 release.
https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/source/geomutils/src/gjk/GuGJKType.h#L154

Pierre-Terdiman, 2020-01-17:
I'll have to double-check at work but I guess I only submitted the fix to the 5.0 trunk.

samvv, 2020-03-17:
Getting this as a hard error on Clang 10, together with the following error (which I turned into a warning):

```
/opt/PhysX-4.1/physx/source/foundation/include/PsSort.h:65:2: warning: use of function '__builtin_alloca' is discouraged; there is no way to check for failure but failure may still occur, resulting in a possibly exploitable security vulnerability [-Walloca]
```
I'm presuming this will be fixed in the next release, but I thought I'd let you know just to be sure.

PierreTerdiman, 2021-04-09:
Our usage of alloca is safe, it goes through the PX_ALLOCA macro which only uses alloca for sizes smaller than 1Kb. We are not going to introduce more runtime memory allocations just to please compilers that do not recognize the code is safe.

closed 2021-04-09 by PierreTerdiman