NVIDIAGameWorks/PhysX #166: Question about TriangleMesh collision problem

utilForever, 2019-08-19:
Hello, I have a question about TriangleMesh collision in PhysX. I created 4 PxShapes using PxTriangleMesh for displacement. This shapes have same y-value. In other words, The thickness of this shape is 0.
When I created dynamic prop that is created using PxConvexShape, it fell down ignoring triangle meshes Of course, I could see that onContact function is called. However, convex meshes surrounding the world is well handled collision process. What should I do to solve this problem?

This is screenshot.
[image]

And this is PXD file.
[Problem.zip](https://github.com/NVIDIAGameWorks/PhysX/files/3514896/Problem.zip)

Pierre-Terdiman, 2019-08-19:
(FWIW we had a thread about this on Twitter and it seems to be fixed now - https://twitter.com/utilforever/status/1163350180692815873?s=20)

utilForever, 2019-08-19:
@Pierre-Terdiman Thanks! 🎆

closed 2019-08-19 by utilForever