NVIDIAGameWorks/PhysX #168: Question about collision filter shader

utilForever, 2019-08-23:
Hello. I have an another question. In collision filter shader, is there a way to get a pxShape of two objects detected instead of flags? I need shapes to check that they should collide by calling another function. BTW, how can I handle the filter callback such as eNOTIFY_TOUCH_LOST and eKILL | eNOTIFY to ignore collisions?

Pierre-Terdiman, 2019-08-23:
You need to use a filter callback if you want the PxShape pointers.

I'm not sure I understand the second part of the question. If you want to ignore collisions, you can return PxFilterFlag::eKILL from the filter callback.

utilForever, 2019-08-23:
Isn't filter callback referring to functions like `onContact`? Can you show me an example code?

Pierre-Terdiman, 2019-08-23:
No, onContact is part of the PxSimulationEventCallback. This is not the same as PxSimulationFilterCallback.

Maybe try this new snippet that I tweeted about recently:

https://twitter.com/PierreTerdiman/status/1162003837886054400?s=20

Some of the scenes there use a filter callback.

utilForever, 2019-08-23:
Thanks! I'll check and apply it to my code. :+1:

utilForever, 2019-08-26:
I changed my code like below. 

```
physx::PxFilterFlags CollisionFilterShader(
	physx::PxFilterObjectAttributes attributes0, physx::PxFilterData filterData0,
	physx::PxFilterObjectAttributes attributes1, physx::PxFilterData filterData1,
	physx::PxPairFlags& retPairFlags, const void* /*constantBlock*/, physx::PxU32 /*constantBlockSize*/)
{
	const physx::PxU32 needsReport = physx::PxU32(GROUND_FLAG | PROP_FLAG);
	physx::PxU32 flags = (filterData0.word3 | filterData1.word3);

	//retPairFlags = physx::PxPairFlag::eCONTACT_DEFAULT;
	//retPairFlags |= physx::PxPairFlag::eDETECT_DISCRETE_CONTACT;
	//retPairFlags |= physx::PxPairFlag::eDETECT_CCD_CONTACT;

	retPairFlags = physx::PxPairFlag::eCONTACT_DEFAULT;
	retPairFlags |= physx::PxPairFlag::eDETECT_CCD_CONTACT | physx::PxPairFlag::eNOTIFY_TOUCH_CCD;

	//if ((flags & needsReport) == needsReport)
	//{
	//	retPairFlags |= (physx::PxPairFlag::eNOTIFY_TOUCH_FOUND | physx::PxPairFlag::eNOTIFY_CONTACT_POINTS);
	//	//retPairFlags |= physx::PxPairFlag::eMODIFY_CONTACTS;
	//}

	//return physx::PxFilterFlag::eDEFAULT;
	return physx::PxFilterFlag::eCALLBACK;
}

	virtual	physx::PxFilterFlags pairFound(physx::PxU32 /*pairID*/,
		physx::PxFilterObjectAttributes /*attributes0*/, physx::PxFilterData /*filterData0*/, const physx::PxActor* /*a0*/, const physx::PxShape* s0,
		physx::PxFilterObjectAttributes /*attributes1*/, physx::PxFilterData /*filterData1*/, const physx::PxActor* /*a1*/, const physx::PxShape* s1,
		physx::PxPairFlags& pairFlags)
	{
		if (s0->userData || s1->userData)
		{
			auto pObject0 = ((CPhysCollideUserData*)s0->userData)->physObject;
			auto pObject1 = ((CPhysCollideUserData*)s1->userData)->physObject;

			if (m_pSolver && !m_pSolver->ShouldCollide(pObject0, pObject1, pObject0->GetGameData(), pObject1->GetGameData()))
			{
				pairFlags = physx::PxPairFlag::eNOTIFY_TOUCH_LOST;
				return physx::PxFilterFlag::eKILL;
			}
		}

		pairFlags = physx::PxPairFlag::eCONTACT_DEFAULT;
		pairFlags |= physx::PxPairFlag::eDETECT_CCD_CONTACT | physx::PxPairFlag::eNOTIFY_TOUCH_CCD;

		return physx::PxFilterFlags();
	}
```
However, this code prints "assertion failed" in debug build.
```
Expression: ...\physx_3.4\source\simulationcontroller\src\ScActorPair.h(150) : Assertion failed: aPair.isReportPair()
```

Pierre-Terdiman, 2019-08-26:
I am unable to reproduce this problem. I tried moving two objects in contact and then going through the disable-collision codepath in the callback, and it worked as expected. Not sure what to do to repro this issue....