NVIDIAGameWorks/PhysX #169: Is it possible to add/remove shapes from actor using PxBVHStructure?

Zeblote, 2019-08-25:
Assume I have an actor that contains 10000 shapes and was initialized with a PxBVHStructure. I now want to add another shape to this actor. Is using the normal attachShape / detachShape methods the correct way to do that while keeping the scene query in actor-centric mode?

AlesBorovicka, 2019-08-26:
Hi,
 yes that is correct, you can use the regular attach/detachShape it will add the shape into the compound pruner only. The rebuild of the main tree should not be triggered. The PxBVHStructure data are used only during the create time, then internally incremental AABB tree is used. So if shapes are added/removed during simulation the tree does remove/add objects accordingly. 

Regards,
Ales

Zeblote, 2019-08-26:
Thanks! That's great to know.

closed 2019-08-26 by Zeblote