NVIDIAGameWorks/PhysX #170: clang issue on make sdk

MeoliNasa, 2019-08-26:
When I `make` the linux-checked version (Ubuntu 18.04 LTS), im getting the following error: 

```
[ 56%] Linking CXX executable /home/rob/Downloads/PhysX-4.1/physx/bin/linux.clang/checked/SnippetImmediateMode_64

/usr/bin/ld: cannot find -lX11

clang: error: linker command failed with exit code 1 (use -v to see invocation)

sdk_snippets_bin/CMakeFiles/SnippetImmediateMode.dir/build.make:130: recipe for target '/home/rob/Downloads/PhysX-4.1/physx/bin/linux.clang/checked/SnippetImmediateMode_64' failed

make[2]: *** [/home/rob/Downloads/PhysX-4.1/physx/bin/linux.clang/checked
/SnippetImmediateMode_64] Error 1

CMakeFiles/Makefile2:641: recipe for target 'sdk_snippets_bin/CMakeFiles/SnippetImmediateMode.dir/all' failed

make[1]: *** [sdk_snippets_bin/CMakeFiles/SnippetImmediateMode.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

```

What should i do to fix this?

AlesBorovicka, 2019-08-27:
Yes, you need to install a few development packages, they are listed here:
https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/documentation/platformreadme/linux/readme_linux.html

Though those are mainly for x86 target. You might also miss:
sudo apt-get install libx11-dev

MeoliNasa, 2019-08-27:
Thank you very much and sorry for the trivial question!

closed 2019-08-27 by MeoliNasa