NVIDIAGameWorks/PhysX #173: PhysXGpu_64.dll

WWebber, 2019-08-27:
Is there any reason, why I cannot find the PhysXGpu_64.dll in the "win.x86_64.vc141.mt" path ?
Was it forgotten to get copied from the installer?

AlesBorovicka, 2019-08-27:
Yes, we distribute only PhysXGpu_64.dll in in the vc140 dir (vc120 for the x86), its then copied during generate_project.bat to appropriate folders (vc141, vc120 etc)

WWebber, 2019-08-31:
Ok. Thx. I already recognized, it's always the same version for all platforms.
BTW: What's the 'generate_project.bat' procedure to build V141 under VS2019? I cannot create the V141 builds, w/o having VS2017 installed?

AlesBorovicka, 2019-09-01:
Yes, that is correct. The projects are generated using cmake and cmake does check for compiler before creating those projects. So if you dont have the compiler, then the project generate fails. For VS2019 you want to user the vc16win64 preset.

WWebber, 2019-09-01:
That's what I did, but I have to build against V141 for integration in 3dsmax plugins. I don't know if it really matters using V142, but I don't want to find it out. Had to execute some find/replace tasks, now I have it working properly. I suggest adding an extra option to 'generate_project.bat'..