NVIDIAGameWorks/PhysX #179: [help] MDd_DynamicDebug' doesn't match value 'MD_DynamicRelease'

UnidayStudio, 2019-09-06:
I'm trying to build PhysX with my app following [this](https://gameworksdocs.nvidia.com/PhysX/4.0/documentation/PhysXGuide/Manual/BuildingWithPhysX.html) guide, but I'm getting the following errors when I compile my project:
```code
1>FastEngine.lib(PhysicsSystem.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '2' doesn't match value '0' in main.obj
1>FastEngine.lib(PhysicsSystem.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MDd_DynamicDebug' doesn't match value 'MD_DynamicRelease' in main.obj
```

I don't know exactly what I missed but looks like PhysX keeps compiling something in debug mode

My project solutions and PhysX where compiled in Release x64 mode and the runtime library is set to Multi-threaded DLL (/MD).

Here is the config XML file that I used to generate the solution:
```xml
<?xml version="1.0" encoding="utf-8"?>
<preset name="vc15win64" comment="VC15 Win64 PhysX general settings">
  <platform targetPlatform="win64" compiler="vc15" />
  <CMakeSwitches>
    <cmakeSwitch name="PX_BUILDSNIPPETS" value="True" comment="Generate the snippets" />
    <cmakeSwitch name="PX_BUILDPUBLICSAMPLES" value="False" comment="Generate the samples projects" />
    <cmakeSwitch name="PX_GENERATE_STATIC_LIBRARIES" value="False" comment="Generate static libraries" />
    <cmakeSwitch name="NV_USE_STATIC_WINCRT" value="False" comment="Use the statically linked windows CRT" />
    <cmakeSwitch name="NV_USE_DEBUG_WINCRT" value="False" comment="Use the debug version of the CRT" />
    <cmakeSwitch name="PX_FLOAT_POINT_PRECISE_MATH" value="False" comment="Float point precise math" />
  </CMakeSwitches>
  <CMakeParams>
    <cmakeParam name="CMAKE_INSTALL_PREFIX" value="install/vc15win64/PhysX" comment="Install path relative to PhysX SDK root" />
  </CMakeParams>
</preset>```

Can you help me?

AlesBorovicka, 2019-09-06:
Hi,
 not sure how this is related to PhysX compilation? From the link error I dont see anything related to PhysX SDK,

UnidayStudio, 2019-09-06:
@AlesBorovicka I don't know where to change if I want to build PhysX in MDd_DynamicDebug, MT_StaticRelease or MD_DynamicRelease. If I go to Code Generation > Runtime LIbrary it changes nothing.

AlesBorovicka, 2019-09-06:
Here you define if debug CRT is used:

alanjfs, 2019-09-14:
Wow. "How to lose a user in 1 comment" - Written and Directed by AlesBorovicka. I have a similar issue and ended up here after a search on the interwebs.

alanjfs, 2019-09-16:
@UnidayStudio You probably solved this by now, but for future readers, I managed to solve this via a related issue here: https://github.com/NVIDIAGameWorks/PhysX/issues/62

In a nutshell, and what Ales was trying to say about was that you can edit the template XML file the [`generate_projects.bat`](https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/generate_projects.bat) file uses to build a Dynamic release instead of the default Static.

- See [this line](https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/buildtools/presets/public/vc16win64.xml#L9) specifically

Set this to `False` and re-build the solution. The resulting build should say `win.x86_64.vc142.md` instead of the default `win.x86_64.vc142.mt`.

alanjfs, 2019-11-05:
Ended up referring back to this issue after having forgotten how to do it. So, for yours and my future selves copy/pasting pleasure:

**physx\buildtools\presets\public\vc14win64_mdd.xml**

```bat
<?xml version="1.0" encoding="utf-8"?>
<preset name="vc14win64_mdd" comment="VC14 Win64 PhysX general settings">
  <platform targetPlatform="win64" compiler="vc14" />
  <CMakeSwitches>
    <cmakeSwitch name="PX_BUILDSNIPPETS" value="False" comment="Generate the snippets" />
    <cmakeSwitch name="PX_BUILDPUBLICSAMPLES" value="False" comment="Generate the samples projects" />
    <cmakeSwitch name="PX_GENERATE_STATIC_LIBRARIES" value="False" comment="Generate static libraries" />
    <cmakeSwitch name="NV_USE_STATIC_WINCRT" value="False" comment="Use the statically linked windows CRT" />
    <cmakeSwitch name="NV_USE_DEBUG_WINCRT" value="True" comment="Use the debug version of the CRT" />
    <cmakeSwitch name="PX_FLOAT_POINT_PRECISE_MATH" value="False" comment="Float point precise math" />
  </CMakeSwitches>
  <CMakeParams>
    <cmakeParam name="CMAKE_INSTALL_PREFIX" value="install/vc14win64_mdd/PhysX" comment="Install path relative to PhysX SDK root" />
  </CMakeParams>
</preset>
```

Note this also sets examples and snippets to `False`, to speed up build times.

Pacheco95, 2022-10-17:
I'm facing the same problem with static linking
https://github.com/NVIDIAGameWorks/PhysX/issues/115#issuecomment-1280115308