NVIDIAGameWorks/PhysX #184: How to enable GPU reduce coordinate articulation

arpspoof, 2019-09-21:
Hi,

I'm trying to simulate my reduced coordinate robot on GPU however I did not find a specific tutorial for that. So I'm following instructions in https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/GPURigidBodies.html and also code in snippetGRB. However, when GPU dynamics is enabled, my scene (with articulation) gets extremely slow. The simulation will freeze after several frames and then crash. Also, if I just enable GPU broad phase, the scene will not experiencing problems but still gets much slower than before. It's even slower than CPU simulation with 0 worker threads. I'm working on arch linux however similar issue also happens on windows.

Am I setting up GPU articulation correctly or is this a potential bug?

Thanks!

WWebber, 2019-09-22:
you forgot to tell us something about the CUDA hardware and drivers you are using..

arpspoof, 2019-09-22:
Hi, the observations on Windows machine is listed in the following:

**Hardware**
Graphics card: GTX 1080

**Cuda**
nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2019 NVIDIA Corporation
Built on Wed_Apr_24_19:11:20_Pacific_Daylight_Time_2019
Cuda compilation tools, release 10.1, V10.1.168

**Control Panel**
In NVIDIA control panel, PhysX is set to use GPU.

**Compiler**
vc15-64bit (VS2017)

**Code used for testing**: (Copied from SnippetGRB to SnippetArticulation and run SnippetArticulation)
```
	PxCudaContextManagerDesc cudaContextManagerDesc;
#ifdef RENDER_SNIPPET
	cudaContextManagerDesc.interopMode = PxCudaInteropMode::OGL_INTEROP;	//Choose interop mode. As the snippets use OGL, we select OGL_INTEROP
																			//when using D3D, cudaContextManagerDesc.graphicsDevice must be set as the graphics device pointer.
#else
	cudaContextManagerDesc.interopMode = PxCudaInteropMode::NO_INTEROP;
#endif

	gCudaContextManager = PxCreateCudaContextManager(*gFoundation, cudaContextManagerDesc, PxGetProfilerCallback());	//Create the CUDA context manager, required for GRB to dispatch CUDA kernels.
	if (gCudaContextManager)
	{
		if (!gCudaContextManager->contextIsValid())
		{
			gCudaContextManager->release();
			gCudaContextManager = NULL;
		}
	}
```
```
	sceneDesc.cudaContextManager = gCudaContextManager;		//Set the CUDA context manager, used by GRB.

	sceneDesc.flags |= PxSceneFlag::eENABLE_GPU_DYNAMICS;	//Enable GPU dynamics - without this enabled, simulation (contact gen and solver) will run on the CPU.
	sceneDesc.flags |= PxSceneFlag::eENABLE_PCM;			//Enable PCM. PCM NP is supported on GPU. Legacy contact gen will fall back to CPU
	sceneDesc.flags |= PxSceneFlag::eENABLE_STABILIZATION;	//Improve solver stability by enabling post-stabilization.
	sceneDesc.broadPhaseType = PxBroadPhaseType::eGPU;		//Enable GPU broad phase. Without this set, broad phase will run on the CPU.
	sceneDesc.gpuMaxNumPartitions = 8;						//Defines the maximum number of partitions used by the solver. Only power-of-2 values are valid. 
														//A value of 8 generally gives best balance between performance and stability.

```

**Observed**:
Simulation freeze at the very beginning, no frame is rendered, window becomes unresponsive. No console error or warning message. 

If comment out the line for enabling GPU dynamics, then simulation can run, but seems slow..

Thanks for any help, it further information is required, please let me know.

arpspoof, 2019-09-22:
I tried the above test code on my arch linux machine again. Renderer gets a black screen, simulation freeze immediately and becomes unresponsive. Also, get the following error message:

/buildAgent/work/9b8aa436d5e8ada1/source/gpusolver/src/PxgTGSCudaSolverCore.cpp (2189) : internal error : GPU integrateCoreParallel fail to launch kernel!!

/buildAgent/work/9b8aa436d5e8ada1/buildtools/../source/gpucommon/include/PxgCudaUtils.h (53) : internal error : SynchronizeStreams failed

/buildAgent/work/9b8aa436d5e8ada1/buildtools/../source/gpucommon/include/PxgCudaUtils.h (53) : internal error : SynchronizeStreams failed

**Hardware**:
GTX 1080 Ti

**Cuda**:
nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2019 NVIDIA Corporation
Built on Sun_Jul_28_19:07:16_PDT_2019
Cuda compilation tools, release 10.1, V10.1.243

**Compiler**:
clang++ --version
clang version 8.0.1 (tags/RELEASE_801/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin

kstorey-nvidia, 2019-09-22:
Hi

GPU TGS solver support with articulations is experimental. There were some incomplete aspects of the implementation in the most recent release. The default (PGS) solver should be completely functional.

We are working on completing the TGS implementation and expect it to be ready very soon. 

Regarding performance, it is not surprising if performance is slower in simpler scenes. It should perform better in larger scenes. If you take a look at the GPU rigid body snippet, it simulates around 8000 rigid bodies. The threshold where GPU begins to beat the CPU is usually around the 1000 rigid body mark, depending on the GPU and CPU being used. If you just simulate a small number of actors, CPU will usually outperform GPU.

Thanks

Kier

arpspoof, 2019-09-22:
Thanks for the quick response!

I tried the PGS solver and it works. No more errors appear. However, when I try to add joint forces via joint cache (in my own robot case), the simulation breaks immediately. Lots of NaN show up in joint positions and velocities. Is this a known issue?

Thanks!

kstorey-nvidia, 2019-09-22:
It should work equivalently to the CPU simulation. Could you please post a code snippet reproducing this issue and we can investigate.

arpspoof, 2019-09-22:
I've done some pre-investigation for you. Looks like the problem is not really related to joint cache but related to inverse dynamics. If I call commonInit() before simulation, the simulation will be broken by bunch of NaN values even if I do not touch joint cache at all. 

I believe this should be clear for further investigation on your side. As for a concrete example, just enable GPU articulation in SnippetArticulation and call commonInit before stepping simulation. Then things will break.

kstorey-nvidia, 2019-09-23:
Thanks. I'll try to reproduce this locally.

lasagnaphil, 2020-04-13:
I am also having this issue in GPU articulations. In fact, even if I don't call articulation->commonInit() at all, the joint cache values will have a bunch of NaN values if articulation->applyCache() is called before the first Scene::simulate(). (This problem does not appear on CPU)

Also, it seems that the root velocity is not being reset with articulation->applyCache(), even when using eROOT and eVELOCITY flags explicitly. Here's an GIF example (This is when I'm setting everything in the articulation to zero every frame after simulate(), but the root velocity doesn't reset to zero somehow):

[image]

It seems that there are some weird behaviors in PxArticulationCache when using the GPU. Any thoughts on why this is happening?