NVIDIAGameWorks/PhysX #186: Cross compile for linux on windows

Zeblote, 2019-09-22:
Is it possible to build linux libs on windows? I noticed various hints suggesting this is possible, such as these lines here in `cmake_generate_projects.py`:
```
winPresetFilter = ['win','uwp','ps4','switch','xboxone','android','crosscompile']
```
```
if self.compiler == 'clang-crosscompile':
```
However, the `crosscompile` preset doesn't exist in the public folder. Am I missing something simple?

AlesBorovicka, 2019-09-23:
Yes, we dont include the linux-crosscompile preset. The preset could look like this:
linux-crosscompile.xml:
`
<?xml version="1.0" encoding="utf-8"?>
<preset name="linux-crosscompile" comment="Linux cross compilation preset, without GPU support">
  <platform targetPlatform="linux" compiler="clang-crosscompile" />
  <CMakeSwitches>
    <cmakeSwitch name="PX_GENERATE_STATIC_LIBRARIES" value="True" comment="Generate static libraries" />
  </CMakeSwitches>
  <CMakeParams>
    <cmakeParam name="CMAKE_INSTALL_PREFIX" value="install/linux-crosscompile/PhysX" comment="Install path relative to PhysX SDK root" />
  </CMakeParams>
</preset>`

But note, that linux cross compile toolchain will be used, it is located in the PhysX/extenals/cmakemodules/linux
`            if self.compiler == 'clang-crosscompile':
                outString = outString + ' -DCMAKE_TOOLCHAIN_FILE=' + \
                    os.environ['PM_CMakeModules_PATH'] + \
                    '/linux/LinuxCrossToolchain.x86_64-unknown-linux-gnu.cmake'
`
In the toolchain file you would have to set your cross compile directory, as we dont include that on github. See this line in LinuxCrossToolchain.x86_64-unknown-linux-gnu.cmake
`SET(LINUX_ROOT $ENV{PM_PACKAGES_ROOT}/clang-crosscompile/5.0.0/x86_64-unknown-linux-gnu)`

Zeblote, 2019-09-23:
Thanks! That worked perfectly.

closed 2019-09-23 by Zeblote