NVIDIAGameWorks/PhysX #193: PxDistanceJoint understanding problems

WWebber, 2019-10-02:
I'd like to use a distance joint with hard-limits, i.e it can only move within a certain range and within the range, the spring should apply a force. But when reading the doc, the PxDistanceJoint does exactly the opposite. The spring is only active outside the given range and I cannot define a certain working range?

alanjfs, 2019-10-15:
Which part of the documentation are you reading? Have you tried [this](https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxapi/files/classPxDistanceJoint.html#aa1f0551bbf0f5cac9878e41525d67e9a)? It sounds like you've set a minimum distance, when what you're looking for it to set the maximum distance.

WWebber, 2019-10-15:
No, I mean this: [https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxapi/files/classPxDistanceJoint.html#a2ac2f3e6b8afe91d7b39e34e34310e28](url)
"The spring is used if enabled, and the distance exceeds the range [min-error, max+error]."
I understand it that way, that only when the MaxDistance is exceeded, the spring will be active?
I'd like to have a spring within a certain limtted range.

alanjfs, 2019-10-16:
> I'd like to have a spring within a certain limtted range.

Aah, yes. I think you've got it right, in that it only activates past the limit, like the D6 and hinge joints do.

Just so I understand what you're looking for, you want:

1. From 0-1, actively pull the bodies towards each other, at a given stiffness
2. From 1+, to stop entirely; like an inverted sphere collider

If so, could you use two distance constraints? One with an active but small max distance, and another with a large min distance. The former being weak, the latter strong.

WWebber, 2019-10-16:
Now you got it! The idea with an additional dist-joint sounds great to me, I'll check it.
My general problem is (btw. since Novodex), the more joints I attach to an actor, making the simulation stable again by juggling around with mass, damping, stiffness,.. always costs me days.

alanjfs, 2019-10-17:
Let me know how it goes, also curious. :)