NVIDIAGameWorks/PhysX #194: Build Failed with latest version of Physx and Xcode

evandarcy, 2019-10-03:
Hi,

I cloned the repo today and have been trying to build Physx from the command line using `xcodebuild -project PhysXSDK.xcodeproj -alltargets -configuration debug`

When I run the command I am greeted with the following error message.

[image]

I am hoping somebody has run into similar issues because I am at a loss for now unfortunately. I'm using OSX on Mojave.

AlesBorovicka, 2019-10-03:
Hi,
 there was a merge request that disabled this warning:
https://github.com/NVIDIAGameWorks/PhysX/pull/185

evandarcy, 2019-10-03:
Thanks for the reply. I cloned the repo just a few hours ago and got that error. Is that merge request on a separate branch or has it been merged with the master branch?

AlesBorovicka, 2019-10-03:
Its merged into our main line, but has not yet been released, so I suggest to apply that fix locally.

evandarcy, 2019-10-03:
Yep, sorted, thanks!

closed 2019-10-03 by evandarcy