NVIDIAGameWorks/PhysX #197: PxContactSet::ignore doesn't work

cmp-, 2019-10-15:
It sets max impulse to 0 but not the eHAS_MAX_IMPULSE flag on the contact patch (like setMaxImpulse does), so the default max impulse is used instead and the contact is not ignored.

Pierre-Terdiman, 2019-10-16:
Thanks, we'll look into it. As a workaround, you can try to set the separation to a high value like 100000.0f, that should effectively ignore the contact as well.

cmp-, 2019-10-16:
setMaxImpulse(i, 0) works just fine for that (and is more efficient since there's an explicit check for 0 impulse that discards the contact early on). I'm pretty sure ignore(i) is supposed to be equivalent to setMaxImpulse(i, 0), just right now it's missing that flag.

Pierre-Terdiman, 2019-10-16:
Yes setMaxImpulse(i, 0) works too and I think you're right about what happened here (ignore() is missing that flag)

closed 2020-03-24 by PierreTerdiman