NVIDIAGameWorks/PhysX #200: Code cleanup

ValZapod, 2019-10-17:
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/samples/pxtoolkit/src/PxTkBmpLoader.cpp#L166
Where is free(line);
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/samples/sampleframework/renderer/src/Renderer.cpp#L940
delete data;
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/geomutils/src/mesh/GuMidphaseRTree.cpp#L380
Strange
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/scenequery/src/SqBucketPruner.cpp#L2053
Unite case PxGeometryType::eCAPSULE: and case PxGeometryType::eCONVEXMESH:
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/samples/sampleframework/platform/src/linux/LinuxSamplePlatform.cpp#L502
Unite cases
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/samples/sampleframework/renderer/src/ogl/OGLRendererTexture2D.cpp#L60
And these cases as well
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/lowleveldynamics/src/DyFeatherstoneArticulation.cpp#L2168
Should not this be in `if (unit.rigidCore1)`

https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/lowleveldynamics/include/DyFeatherstoneArticulationJointData.h#L162
delete this line
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/lowlevelaabb/src/BpBroadPhaseSapAux.cpp#L121
delete this line or 124
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/simulationcontroller/src/ScNPhaseCore.h#L179
delete mCurrentOffset = (see line 186)
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/lowleveldynamics/src/DyTGSDynamics.cpp#L1145-L1146
Delete this lines 
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/snippets/snippetvehiclenodrive/SnippetVehicleNoDrive.cpp#L142
Delete this line and
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/snippets/snippetvehiclemultithreading/SnippetVehicleMultiThreading.cpp#L212
this one and
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/snippets/snippetvehicle4w/SnippetVehicle4W.cpp#L196
this one and
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/snippets/snippetvehicletank/SnippetVehicleTank.cpp#L180
this one and 
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/snippets/snippetvehiclescale/SnippetVehicleScale.cpp#L206
this one.
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/samples/sampleframework/renderer/src/ogl/OGLRenderer.cpp#L598-L599
Can we unite those two lines?
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/physx/src/buffering/ScbBody.h#L836-L838
You can delete `(!mBufferedLinVelocity.isZero()) && ` The same on line 854
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/simulationcontroller/src/ScNPhaseCore.cpp#L293-L296
The '||' operator is surrounded by opposite expressions
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/lowleveldynamics/src/DyFeatherstoneArticulation.cpp#L2510
What is the point to check for j.motion[i] != PxArticulationMotion::eLOCKED The same on line 2666

Do not close the issue, it is not all <3

ValZapod, 2019-10-17:
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/simulationcontroller/src/ScScene.cpp#L2441
Comment this))
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/lowlevel/common/src/pipeline/PxcNpBatch.cpp#L232
If buffer.count were zero we would have already returned on line 228. The same 
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/lowlevel/common/src/pipeline/PxcNpBatch.cpp#L249
here 
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/lowlevel/common/src/pipeline/PxcNpBatch.cpp#L255
here 
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/lowlevel/common/src/pipeline/PxcNpBatch.cpp#L267
and here.
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/pvd/src/PxPvdDefaultFileTransport.cpp#L172
return true
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/pvd/src/PxPvdObjectModelMetaData.cpp#L854
Again would've returned if propDType->mRequiresDestruction is true so it is false and cls->mRequiresDestruction never checked
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/physxvehicle/src/PxVehicleComponents.cpp#L63
mNbRatios>=1 maybe >0
the same here https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/physxvehicle/src/PxVehicleDrive.cpp#L76
always false

Pierre-Terdiman, 2019-10-18:
Thanks, I submitted some of these to our trunk. Will appear in next update. Didn't have time to go through all of them yet.

ValZapod, 2019-10-19:
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/physxextensions/src/serialization/Xml/SnXmlVisitorReader.h#L273
No need for this 
https://github.com/NVIDIAGameWorks/PhysX/blob/4050bbfdc2699dfab7edbf0393df8ff96bbe06c5/physx/source/geomutils/src/GuAABBTreeBuild.cpp#L218
This is VERY strange. Check line 180. Both variables are not changed in between.
https://github.com/NVIDIAGameWorks/PhysX/blob/4050bbfdc2699dfab7edbf0393df8ff96bbe06c5/physx/source/geomutils/src/contact/GuContactPolygonPolygon.cpp#L712
What is the point?
https://github.com/NVIDIAGameWorks/PhysX/blob/4050bbfdc2699dfab7edbf0393df8ff96bbe06c5/physx/source/geomutils/src/sweep/GuSweepCapsuleTriangle.cpp#L312
Again we would've return
https://github.com/NVIDIAGameWorks/PhysX/blob/4050bbfdc2699dfab7edbf0393df8ff96bbe06c5/physx/source/physxcooking/src/Adjacencies.cpp#L684-L686
https://github.com/NVIDIAGameWorks/PhysX/blob/4050bbfdc2699dfab7edbf0393df8ff96bbe06c5/physx/source/physxcooking/src/mesh/TriangleMeshBuilder.cpp#L320-L322
The '(A && B) || (!A && !B)' expression is equivalent to the 'bool(A) == bool(B)' expression.
https://github.com/NVIDIAGameWorks/PhysX/blob/4050bbfdc2699dfab7edbf0393df8ff96bbe06c5/physx/source/physxcooking/src/mesh/TriangleMeshBuilder.cpp#L722
What is the point? Maybe it should be line 690?
https://github.com/NVIDIAGameWorks/PhysX/blob/4050bbfdc2699dfab7edbf0393df8ff96bbe06c5/physx/source/common/src/CmPhysXCommon.h#L62
PX_CHECK_AND_RETURN macro checks for exp twice, maybe there is some other way?
https://github.com/NVIDIAGameWorks/PhysX/blob/4050bbfdc2699dfab7edbf0393df8ff96bbe06c5/physx/source/physxextensions/src/ExtSimpleFactory.cpp#L103
No need to check shape
https://github.com/NVIDIAGameWorks/PhysX/blob/4050bbfdc2699dfab7edbf0393df8ff96bbe06c5/physx/source/physx/src/NpPhysics.cpp#L244
Should be that before?
https://github.com/NVIDIAGameWorks/PhysX/blob/4050bbfdc2699dfab7edbf0393df8ff96bbe06c5/physx/source/physxextensions/src/serialization/Xml/SnRepXCoreSerializer.cpp#L63
What is the point?
https://github.com/NVIDIAGameWorks/PhysX/blob/4050bbfdc2699dfab7edbf0393df8ff96bbe06c5/physx/source/physxextensions/src/serialization/Xml/SnRepXUpgrader.cpp#L328
https://github.com/NVIDIAGameWorks/PhysX/blob/4050bbfdc2699dfab7edbf0393df8ff96bbe06c5/physx/source/physxextensions/src/serialization/Xml/SnRepXUpgrader.cpp#L404
Exessive. substring "PxCloth" is enough.
https://github.com/NVIDIAGameWorks/PhysX/blob/4050bbfdc2699dfab7edbf0393df8ff96bbe06c5/physx/snippets/snippetvehiclecommon/SnippetVehicleCreate.cpp#L247
delete line
https://github.com/NVIDIAGameWorks/PhysX/blob/4050bbfdc2699dfab7edbf0393df8ff96bbe06c5/physx/samples/pxtoolkit/src/PxTkBmpLoader.cpp#L132-L136
Wow, put it in one if, would you?
https://github.com/NVIDIAGameWorks/PhysX/blob/4050bbfdc2699dfab7edbf0393df8ff96bbe06c5/physx/snippets/snippetcontactreportccd/SnippetContactReportCCD.cpp#L230
Should not this check be before gSphereActor was used?
./physx/samples/samplebase/AcclaimLoader.cpp	116	err	V739 EOF should not be compared with a value of the 'char' type. The 'c' should be of the 'int' type.
./physx/samples/samplebase/AcclaimLoader.cpp	132	err	V739 EOF should not be compared with a value of the 'char' type. The 'c' should be of the 'int' type.
./physx/samples/samplebase/AcclaimLoader.cpp	144	err	V739 EOF should not be compared with a value of the 'char' type. The 'c' should be of the 'int' type.
./physx/samples/samplebase/AcclaimLoader.cpp	167	err	V739 EOF should not be compared with a value of the 'char' type. The 'c' should be of the 'int' type.
./physx/samples/samplebase/AcclaimLoader.cpp	179	err	V739 EOF should not be compared with a value of the 'char' type. The 'c' should be of the 'int' type.
./physx/samples/samplebase/AcclaimLoader.cpp	202	err	V739 EOF should not be compared with a value of the 'char' type. The 'c' should be of the 'int' type.
./physx/samples/samplebase/AcclaimLoader.cpp	214	err	V739 EOF should not be compared with a value of the 'char' type. The 'c' should be of the 'int' type.
./physx/samples/samplebase/AcclaimLoader.cpp	226	err	V739 EOF should not be compared with a value of the 'char' type. The 'c' should be of the 'int' type.

https://www.viva64.com/en/w/v739/

ValZapod, 2019-10-21:
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/samples/samplesubmarine/SampleSubmarine.cpp#L791-L793
delete crab;
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/samples/samplevehicle/SampleVehicle.cpp#L1915-L1922
Do just else not else if
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/samples/sampleframework/renderer/src/ogl/OGLRendererTexture2D.cpp#L129-L130
unite cases
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/samples/sampleframework/renderer/src/ogl/OGLRendererTexture2D.cpp#L46-L47
unite cases
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/samples/sampleframework/renderer/src/ogl/OGLRendererMesh.cpp#L44
Unite cases (line 49)
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/samples/sampleframework/renderer/src/ogl/OGLRendererMaterial.cpp#L85-L87
Unite cases 85 + 87
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/physx/src/buffering/ScbScene.cpp#L1142-L1143
Unite cases
https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/pvd/src/PxPvdObjectModelMetaData.h#L191
Maybe this should be somethimh else? There is alreaady the same function on line 200