NVIDIAGameWorks/PhysX #201: model just single direction has collision

pkxpp, 2019-10-18:
if don't change the model, any other solutions?

[image]
[image]

WWebber, 2019-10-18:
honestly, I don't understand your question.. what shall collide or not collide with what ?

pkxpp, 2019-10-21:
> honestly, I don't understand your question.. what shall collide or not collide with what ?

Collide with a capsule. The first image is the capsule want go through the model(front->back)，but failed. The second image is the capsule want go through the model(back->front), but successed.

kstorey-nvidia, 2019-10-21:
Are you referring to using a PxTriangleMesh and back-face culling for collision?

There is a flag to flag meshes as double-sided. However, it is generally a better idea to model double-sided objects with a small amount of thickness rather than relying on double-sided polygons. Due to discrete sampled collision detection or the solver failing to converge, an object on one side of your model could protrude through the other side of the model slightly if the fence panel is paper thin.

pkxpp, 2019-10-28:
> Are you referring to using a PxTriangleMesh and back-face culling for collision?
>
> There is a flag to flag meshes as double-sided. However, it is generally a better idea to model double-sided objects [...]

Thanks.