NVIDIAGameWorks/PhysX #202: [Regression] Inconsistent result generated by CRBA and ABA

arpspoof, 2019-10-18:
This issue happens only after the August change is released. I'm considering it as a potential regression bug.

Observation:
Accelerations calculated from CRBA algorithm is different from actual simulated accelerations (calculated by ABA). I'm using a 28 dof robot and the 5 entries of the solved acceleration by CRBA has large difference with actual acceleration. 

Rough step to reproduce:
1. Load a robot into the scene. It's better to have more dofs since I'm observing inconsistency for 5 the entries.
2. At each time step, get mass matrix H, centrifugal and coriolis C, gravity G and external force E from inverse dynamics. Apply a random force tau via joint cache. Solve equation H*qddot  + C + G + E = tau and get qddot as acceleration. Save this acceleration until next time step. 
3. Drive the simulation to the next step and acquire real acceleration from joint cache. Compare this with the saved calculated acceleration. Observe that most of the entries are almost the same except for 5 of them. 

I'm suspecting there might be some inconsistency between the two dynamics algorithm caused by the August update. (If I use the March version, the two results will perfectly match).

kstorey-nvidia, 2019-10-19:
Thanks for reporting. We will investigate ASAP.

arpspoof, 2019-11-01:
Hello, when I tried this experiment on March release, we can solve for all joints' acceleration by CRBA correctly. However, if we attempt to solve base accelerations (floating base), it turns out that we can only solve angular acceleration correctly and the linear part is wrong. I'm not sure but I try to change the following part of PhysX code:

File: DyFeatherstoneInverseDynamic.cpp
Function: FeatherstoneArticulation::getGeneralizedGravityForce
When passing gravity into inverseDynamicFloatingBase, instead of using the negated one, I changed to to the original gravity in function argument list.

After this change, everything solved by CRBA will match exactly with actually simulated result by ABA including root's angular and linear acceleration. Therefore, I guess that gravity direction is messed up in either forward dynamics part or inverse dynamics part. 

Hope that will help the investigation. (Note that the above experiment is done on March version so this issue should be an old one)

kstorey-nvidia, 2019-11-01:
Thanks for the follow-up details