NVIDIAGameWorks/PhysX #203: Crash/memory corruption in physx::Sq::CompoundTree::updateMapping!

Zeblote, 2019-10-21:
I'm trying to add new shapes to an actor that was created with a BVHStructure (which is supported according to https://github.com/NVIDIAGameWorks/PhysX/issues/169). However, on adding the third shape, I'm getting memory corruption.

Some details on my environment:
- Using PhysX 4.1 (the latest commit on this repo)
- Using Unreal Engine 4.23.1
- Both compiled using VS 16.2.5 on Windows 10
- Enabled the Stomp Allocator in Unreal Engine to immediately crash when the memory corruption would occur. Without this, the engine would crash in some random location the next frame. Not sure how to do something like this outside Unreal Engine to make a repro case.

What I am trying to do:

1) Create actor with single shape attached
2) Create bvhstructure for actor
3) Create aggregate
4) Add actor to aggregate with bvhstructure
5) Add actor to scene
6) Some time passes
7) Add more shapes to actor

Steps 1-5 are just taken from the bvhstructure snippet. I don't think I missed anything important here.

```
PxTransform PhysXTransform = U2PTransform(ChunkTransform);
PhysXActor = GPhysXSDK->createRigidStatic(PhysXTransform);

// adds a single shape to the actor
CreateBrickCollisionShapes_AssumesLocked(PhysXActor, Brick);

PxU32 NumBounds = 0;
PxBounds3* Bounds = PxRigidActorExt::getRigidActorShapeLocalBoundsList(*PhysXActor, NumBounds);
check(Bounds);

PxBVHStructureDesc PhysXBVHStructureDesc;
PhysXBVHStructureDesc.bounds.count = NumBounds;
PhysXBVHStructureDesc.bounds.data = Bounds;
PhysXBVHStructureDesc.bounds.stride = sizeof(PxBounds3);

// unnecessarily complicated way to get access to PxCooking
PxCooking* PhysXCooking = GetPhysXCookingModule(true)->GetPhysXCooking()->GetCooking();
check(PhysXCooking);

PxBVHStructure* PhysXBVHStructure = PhysXCooking->createBVHStructure(PhysXBVHStructureDesc, GPhysXSDK->getPhysicsInsertionCallback());

PhysXAggregate = GPhysXSDK->createAggregate(1, false);
PhysXAggregate->addActor(*PhysXActor, PhysXBVHStructure);

PhysScene->GetPxScene()->addAggregate(*PhysXAggregate);

PhysXBVHStructure->release();
PhysXBVHStructure = nullptr;

GPhysXAllocator->deallocate(Bounds);
Bounds = nullptr;
```

So far, this works. Now, I try to add more shapes to the actor with attachShape. On the third shape, it attempts to corrupt memory. Interestingly, it works fine to add and remove a shape many times, but when the total amount of shapes in the actor reaches 3, it dies.

Crashes here:
[image]

With callstack:
[image]

Zeblote, 2019-10-21:
Did some investigation: it seems you are never resizing the physx::Sq::CompoundTree::mUpdateMap, so it is accessing the array out of bounds as more shapes get inserted. I added a bodge so the map is initialized with huge space and it solves the crash, but is clearly not a usable solution. (Why doesn't the assert trigger on out of bounds access? Strange.)

AlesBorovicka, 2019-10-21:
Hi,
 Thanks for the report, will take a look asap.

AlesBorovicka, 2019-10-21:
Yes, there is a bug in the code, the update map is indeed not resized. Could you please try to modify the updateMapping code like this:
`	
mPruningPool->addObjects(&result, &bounds, &payload, 1);
if (mPruningPool->mMaxNbObjects > mUpdateMap->size())
		mUpdateMap->resize(mPruningPool->mMaxNbObjects);
`
This should fix the issue, not sure why I did not see this before, sorry about that.

Zeblote, 2019-10-21:
Hmm, that would mean it will never be shrunk if shapes are removed from the compound again, wouldn't it? So the memory usage would remain higher than necessary.

AlesBorovicka, 2019-10-21:
Yes, that is correct. Resizing each time for the exact size does not make much sense to me, there would have to be several mem copies per add/remove.

Zeblote, 2019-10-21:
Oh I don't mean to resize each time of course, but perhaps both the pruning pool and this could be resized back down once only half the space is used

AlesBorovicka, 2019-10-21:
I see, sure I can extend the remove part.

Zeblote, 2019-10-21:
This solves the crash though, thanks!