NVIDIAGameWorks/PhysX #205: Can I disable AggAggPairs between static actors?

Zeblote, 2019-10-21:
In a scene with ~1000 compound PxRigidStatic that contain ~1500000 shapes, adding another shape to one of them causes this to show up in the profile:

[image]

Now since these are all static I will never care about any pairs between them, is it possible to remove these tasks? Maybe a flag I missed?

Pierre-Terdiman, 2019-10-22:
It is not possible in current PhysX version. You did not miss a flag.

Improved aggregate filtering (and indeed, a way to "remove these tasks") has already been implemented for the next PhysX version. If needed, we might be able to back-port the change to make it appear earlier on GitHub.

Zeblote, 2019-10-22:
Sounds great, I will wait for the next version then :)

closed 2019-10-22 by Zeblote

Zeblote, 2019-10-24:
I forgot to ask, do you have a rough estimate when this version will be ready? Like, is it going to be a few weeks, or 3 months, or more?