NVIDIAGameWorks/PhysX #208: CMake targets are not exported

diegoferigo, 2019-10-23:
I am checking the best way to link PhysX against targets of my CMake project. The documentation reports the steps to add the include directories and the linked libraries.

However, considering that PhysX has a very good CMake structure, it would be great if the targets were directly exported by CMake.

I found that an export set already exists:

https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/compiler/cmake/linux/CMakeLists.txt#L114-L118

But it is never installed. This means that something similar is missing:

```cmake
INSTALL(
    EXPORT PhysXSDK
    DESTINATION ...
    NAMESPACE PhysX)
```

The final goal is having properly exported targets that transitively carry properties. Exporting the targets, together with a proper package config file, would allow user code to do the following:

```cmake
# Add the install prefix to the CMake search path.
# It would be compatible also if users install outside the PhysX repo folder.
list(APPEND CMAKE_PREFIX_PATH "/path/to/repo/PhysX/install/linux/PhysX"

find_package(PhysX 4.1.1 REQUIRED COMPONENTS PhysX PhysXCommon PhysXCooking PhysXGpu ...)

add_library(MyLibrary SHARED src/mylibrary.cpp)
target_link_library(MyLibrary PhysX::PhysX PhysX::PhysXCommon PhysX::PhysXGpu ...)
```

Note that in this way users do not have to specify anything but the linked imported target. As a bonus user would also get a proper release matching.

Resources:
- https://gitlab.kitware.com/cmake/community/wikis/doc/tutorials/Packaging
- https://gitlab.kitware.com/cmake/community/wikis/doc/tutorials/Exporting-and-Importing-Targets#exporting-from-an-installation-tree
- https://gitlab.kitware.com/cmake/community/wikis/doc/tutorials/How-to-create-a-ProjectConfig.cmake-file
- https://cmake.org/cmake/help/latest/module/CMakePackageConfigHelpers.html

phcerdan, 2019-11-12:
Hi @diegoferigo, the PR #222 provides a PhysXConfig.cmake for the build and install tree,
so `find_package(PhysX)` can be used.
It also removed the requirement of python and the presets for the different platforms. The goal is to facilitate integration of PhysX in third party projects. Any comments on that PR is welcome, and hope it helps!

diegoferigo, 2019-11-13:
Hi @phcerdan, thanks a lot, it looks great! I left few preliminary comments in the PR. I let you know more as soon as I can test it.