NVIDIAGameWorks/PhysX #211: Removing many shapes from large compounds is very slow

Zeblote, 2019-10-27:
I often have to add/remove large numbers of shapes to/from compounds (but without deleting the entire compounds at once). Profiling removing many individual shapes, I noticed that PhysX is wasting 80% of the total time spent inside it just removing or finding things in linked lists. This seems really bad and basically unacceptable!

Basic VS profiler screenshot:
[image]

This one is wasting a huge amount of time:
https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/source/simulationcontroller/src/ScActorSim.cpp#L76

And so is this one (called from RigidCore::removeShapeFromScene):
https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/source/simulationcontroller/src/ScRigidCore.cpp#L40

Are there maybe any plans to fix this in the next version already? How come linked lists (probably better named cache destroyers) are used at all in PhysX?

Could they be replaced with arrays, like how PtrTable::find/realloc is being used the same amount of times yet runs almost 20 times faster? Maybe even caching the indices inside the array.

PierreTerdiman, 2019-10-28:
Could we get some numbers here? How many shapes do you have in a compound and how many do you remove at the same time?

A linked list is not always worse than an array, there's always a trade-off there. This specific one seems to have appeared a long, long time ago. I suspect the rationale was about saving memory and reducing the number of memory allocations, but I cannot really confirm this at the moment.

Yes we could replace it with an array. Ironically for the use-case you mention (removing a random item) the linked list could be faster if it would be a doubly linked list (containing the "previous" pointer as well). It looks like the only reason we iterate from the start of the list in "onElementDetach" is because we miss the previous pointer to close the hole in the list.

The array would be faster to iterate over in several other places in the SDK though, so that is probably a better idea overall. The only drawback is that it would / could increase memory usage for compounds. On the other hand we would save some memory because the "next in actor" member would vanish.

So anyway give me some numbers and I'll start with a benchmark to make sure our changes make a difference.

Zeblote, 2019-10-28:
In the scene I used for profiling, there were ~2k static compounds with a total of ~1.5m shapes, with a maximum of ~9k shapes in any single compound.

To get a more useful reading from the VS profiler for the screenshot above I removed all of the shapes, but the percentage of overheads would be around the same with smaller numbers being removed at once.

Zeblote, 2019-10-29:
I see. Would it be possible to use such a mapping just for shapes that are not shared? Currently I am not using shared shapes at all since that would require all the instances to have the same transform relative to their actor (unless I completely misunderstood something here).

A batched removal function would be very nice. Maybe a batched add function could also be useful?

PierreTerdiman, 2019-10-30:
I suppose it would be an option to simplify the mapping for non-shared shapes, yes. But I'm not sure it's worth it, as we'd still need another solution for shared shapes anyway - that's twice the work and twice the code to write & maintain. But I added that on my list of things to try (which unfortunately keeps growing the more I look at the problem).

You did not misunderstand shared shapes: the local pose is indeed shared as well, that's by design. The use-case is, for example, a simple compound object for a chair or a table. If you have N of these actors in a level, they can all share the same shapes. There was a line to draw between which data is shared and which data is not. We chose to share the local pose because the corresponding data is relatively large (well it's only 7 floats but that's "large" for a shape :)) and thus, worth sharing to save memory.

A batch function for adding shapes would make sense as well, but it wouldn't have the same opportunities as the removal function for perf gains. Generally speaking though, it would make sense to provide batched versions of many other API functions.

Zeblote, 2020-01-21:
@PierreTerdiman I wonder, did you get to try any of those ideas for the next version?

Zeblote, 2021-04-12:
If anyone finds this in the future, I made a series of small modifications a while back to take removing a shape from anywhere in a large aggregate from O(N) to O(1), at the expense of shared shapes (which I do not use) no longer functioning.

https://github.com/Zeblote/PhysX/commit/08d8c9741e196682b1036fc9d9fb56971b1a8a8c
https://github.com/Zeblote/PhysX/commit/ed8209180e4ee77774d8a742f8755521b17dbb1f
https://github.com/Zeblote/PhysX/commit/5d50eec4c1d72ea9484aa257ed76890b2eb7190c