NVIDIAGameWorks/PhysX #212: PhysX 4.1 Character Controller hit report against triggers

ahanganu, 2019-10-29:
Hello,

I have a small question I can't seem to find a lot of information on.

**Is it normal for character controllers to not report collisions with trigger shapes?**

To give a bit more context now. I've created a simple character controller, and simple static sphere shape. If I leave the shape with **PxShapeFlag::eSIMULATION_SHAPE = true** and **PxShapeFlag::eTRIGGER_SHAPE = false** I get the
**onShapeHit(const PxControllerShapeHit& hit)** callback with it. If I just swap the shape flags and make the sphere a trigger I no longer get any callback.

I've seen a solution of creating such "character only triggers" is to use obstacles but I keep wondering if this is the normal behavior or is it I am doing something wrong?

Any tips would be appreciated.

ahanganu, 2019-10-30:
Thank you for pointing that out. 

I guess what I expected is not to have collisions (in term of reactivity, contact response) with them but get the overlap touch callback to use them as triggers for character controllers.

We'll definitely give your suggestion a try as I am currently having triggers which can only be activated by character controllers. Now I'm using normal static actors and just removing them when I get the collision contact but it's not ideal performance-wise.

closed 2019-10-30 by ahanganu

PierreTerdiman, 2019-10-31:
The character controllers create a regular PxActor under the hood, that should get reported by triggers. It's only the CCT module that disables collisions with trigger obstacles, but once the CCT is in a trigger zone its PxActor/PxShape should get reported by the trigger - I think (haven't double-checked).