NVIDIAGameWorks/PhysX #214: Convex Mesh collision, contacts found in 3.4 but not 4.x

jjandersson, 2019-11-04:
Hello, when migrating from 3.4 to 4.0 I'm experiencing some issues with collision detection. 
Working with a convex mesh that is cooked by providing points only, certain areas simply doesn't generate contacts when using 4.0 (tested quickly with 4.1 as well, no result)

I've tried the following things without result, the "holes" in the mesh are still there

- Enable CCD for the whole scene, all pairs 
- Disable mesh cleaning with PxMeshPreprocessingFlag::eDISABLE_CLEAN_MESH
- Using the previous default broad phase algorithm: PxBroadPhaseType::eSAP
- Experimenting with different values for rest_offset / contact_offset

The simplified example I'm posting uses the same mesh, and a bunch of boxes to illustrate where the holes are. Specifically frame **78** shows where some boxes goes through the mesh. Even when boxes are initialized on the surface with close to 0 velocity they still fall through in these very specific locations.

Ideas would be greatly appreciated, thanks

[collision_example.zip](https://github.com/NVIDIAGameWorks/PhysX/files/3803621/collision_example.zip)

kstorey-nvidia, 2019-11-04:
Could you please try PCM collision (PxSceneFlag::eENABLE_PCM)? It seems you have explicitly disabled it.

It does look like contacts are being missed but I don't know what would have caused it. The legacy collision code has not changed since PhysX 3.4. The only possible differences I can think of would be the convex hull cooking code or some tools difference (different build tool or flags being used).

PCM has been the default collision mode in PhysX 3.4 and we generally try to steer people towards the PCM code wherever possible. It would be great if you could try it out and report back whether this resolves the issue.

jjandersson, 2019-11-04:
Yeees, enabling PxSceneFlag::eENABLE_PCM indeed fixes the problem thank you so much! 
We disabled it since it caused some other issues, but if it's prefered to have it enabled we'll try to deal with those issues in a different way. Huge thanks for the quick response.

closed 2019-11-04 by jjandersson