NVIDIAGameWorks/PhysX #218: Correctly compiling win64 with clang

Zeblote, 2019-11-07:
Epic recently added support for compiling Unreal Engine with clang for win64, and it seems to result in all around slightly better code. So it'd be interesting to try this for physx aswell!

Unfortunately I can't figure out how to do it correctly.

I've attempted to use it as explained here (https://gitlab.kitware.com/cmake/cmake/issues/19174#note_558724), by installing the plugin and adding the `-T llvm` to `cmake_generate_projects.py`. This appears to work properly for generating the cmake projects:

```
Using preset xml: buildtools/presets/public/vc16win64-brickadia.xml
Target platform: win64 using compiler: vc16
PM_CMakeModules_PATH: C:/UnrealEngine/Engine/Source/ThirdParty/PhysX4/physx/../externals/CMakeModules
PM_PATHS: C:/UnrealEngine/Engine/Source/ThirdParty/PhysX4/physx/../externals/CMakeModules;C:/UnrealEngine/Engine/Source/ThirdParty/PhysX4/physx/../externals/targa
Cmake: cmake.exe
Not searching for unused variables given on the command line.
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.18362.
-- The C compiler identification is Clang 9.0.0 with MSVC-like command-line
-- The CXX compiler identification is Clang 9.0.0 with MSVC-like command-line
-- Check for working C compiler: C:/Program Files/LLVM/bin/clang-cl.exe
-- Check for working C compiler: C:/Program Files/LLVM/bin/clang-cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files/LLVM/bin/clang-cl.exe
-- Check for working CXX compiler: C:/Program Files/LLVM/bin/clang-cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
PHYSX ROOT C:/UnrealEngine/Engine/Source/ThirdParty/PhysX4/physx
PhysX Build Platform: windows
Using CXX Compiler: C:/Program Files/LLVM/bin/clang-cl.exe
Added PhysX
-- Configuring done
-- Generating done
-- Build files have been written to: C:/UnrealEngine/Engine/Source/ThirdParty/PhysX4/physx/compiler/vc16win64-brickadia
```

But it doesn't compile like that. Guess it's not that simple :(
```
**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.2.5
** Copyright (c) 2019 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

Microsoft Visual Studio 2019 Version 16.0.29306.81.
Copyright (C) Microsoft Corp. All rights reserved.
1>------ Build started: Project: ZERO_CHECK, Configuration: debug x64 ------
1>Checking Build System
2>------ Build started: Project: PhysXFoundation, Configuration: debug x64 ------
3>------ Build started: Project: LowLevel, Configuration: debug x64 ------
4>------ Build started: Project: LowLevelAABB, Configuration: debug x64 ------
5>------ Build started: Project: LowLevelDynamics, Configuration: debug x64 ------
6>------ Build started: Project: PhysXTask, Configuration: debug x64 ------
7>------ Build started: Project: SceneQuery, Configuration: debug x64 ------
8>------ Build started: Project: SimulationController, Configuration: debug x64 ------
9>------ Build started: Project: FastXml, Configuration: debug x64 ------
2>Building Custom Rule C:/UnrealEngine/Engine/Source/ThirdParty/PhysX4/physx/source/compiler/cmake/CMakeLists.txt
9>Building Custom Rule C:/UnrealEngine/Engine/Source/ThirdParty/PhysX4/physx/source/compiler/cmake/CMakeLists.txt
3>Building Custom Rule C:/UnrealEngine/Engine/Source/ThirdParty/PhysX4/physx/source/compiler/cmake/CMakeLists.txt
2>In file included from C:\UnrealEngine\Engine\Source\ThirdParty\PhysX4\physx\source\foundation\src\PsAllocator.cpp:30:
2>In file included from C:\UnrealEngine\Engine\Source\ThirdParty\PhysX4\physx\source\foundation\include\PsFoundation.h:33:
2>In file included from C:\UnrealEngine\Engine\Source\ThirdParty\PhysX4\physx\..\pxshared\include\foundation/PxErrors.h:36:
4>Building Custom Rule C:/UnrealEngine/Engine/Source/ThirdParty/PhysX4/physx/source/compiler/cmake/CMakeLists.txt
2>In file included from C:\UnrealEngine\Engine\Source\ThirdParty\PhysX4\physx\..\pxshared\include\foundation/Px.h:37:
2>In file included from C:\UnrealEngine\Engine\Source\ThirdParty\PhysX4\physx\..\pxshared\include\foundation/PxSimpleTypes.h:40:
2>C:\UnrealEngine\Engine\Source\ThirdParty\PhysX4\physx\..\pxshared\include\foundation/PxPreprocessor.h(35): fatal error : 'ciso646' file not found
5>Building Custom Rule C:/UnrealEngine/Engine/Source/ThirdParty/PhysX4/physx/source/compiler/cmake/CMakeLists.txt
6>Building Custom Rule C:/UnrealEngine/Engine/Source/ThirdParty/PhysX4/physx/source/compiler/cmake/CMakeLists.txt
7>Building Custom Rule C:/UnrealEngine/Engine/Source/ThirdParty/PhysX4/physx/source/compiler/cmake/CMakeLists.txt
9>In file included from C:\UnrealEngine\Engine\Source\ThirdParty\PhysX4\physx\source\fastxml\src\PsFastXml.cpp:30:
9>In file included from C:\UnrealEngine\Engine\Source\ThirdParty\PhysX4\physx\include\foundation/PxAssert.h:33:
9>In file included from C:\UnrealEngine\Engine\Source\ThirdParty\PhysX4\physx\include\foundation/PxFoundationConfig.h:33:
9>C:\UnrealEngine\Engine\Source\ThirdParty\PhysX4\physx\..\pxshared\include\foundation/PxPreprocessor.h(35): fatal error : 'ciso646' file not found
8>Building Custom Rule C:/UnrealEngine/Engine/Source/ThirdParty/PhysX4/physx/source/compiler/cmake/CMakeLists.txt
2>Done building project "PhysXFoundation.vcxproj" -- FAILED.
9>Done building project "FastXml.vcxproj" -- FAILED.
```

Any ideas what I have to do differently for this to actually work?

amoravanszky, 2019-12-27:
I suppose you will have to painstakingly fix all the errors for it to work -- its hard to guess at what the issues will be.  I would expect it to be feasible but I am not surprised it doesn't just work since its not something we have done.  Probably you can use the linux specific code blocks for some of the fixes.

alanjfs, 2020-01-21:
@Zeblote Did you manage to make progress on this? I'm also interested, how about we put together a fork and try to sort out the issues? Anyone else interested in this? I'm moving my projects over to clang-cl, and PhysX is the one hold-up currently.

Zeblote, 2020-01-22:
Unfortunately I didn't get to experiment more with it yet, but I'd definitely like to try it out if you manage to make it work!

alanjfs, 2020-01-22:
> but I'd definitely like to try it out if you manage to make it work!

The offer was for working together to make it work. xD