NVIDIAGameWorks/PhysX #219: Inconsistent __declspec macro between definition and declaration of PxTransformFromSegment

phcerdan, 2019-11-11:
The declaration has the macro `PX_FOUNDATION_API` which expands to: `__declspec(dllimport)`:

https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/include/geometry/PxCapsuleGeometry.h#L114

And the definition `PX_PHYSX_COMMON_API` which expands to `__declspec(dllexport)`:

https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/source/common/src/CmMathUtils.cpp#L59

It generates a warning when building shared (non-static) libraries in `msvc 2019`.
```cpp
LINK : warning LNK4217: symbol '?PxTransformFromSegment@physx@@YA?AVPxTransform@1@AEBVPxVec3@1@0PEAM@Z
(class physx::PxTransform __cdecl physx::PxTransformFromSegment(class physx::PxVec3 const &,class physx::PxVec3 const &,float *))'
defined in 'CmMathUtils.obj' is imported by 'GuMTD.obj'
in function '"bool __cdecl pointConvexDistance(class physx::PxVec3 &,class physx::PxVec3 &,float &,class physx::PxVec3 const &,
             class physx::Gu::ConvexMesh const *,class physx::PxMeshScale const &,class physx::PxTransform const &)"
(?pointConvexDistance@@YA _NAEAVPxVec3@physx@@0AEAMAEBV12@PEBVConvexMesh@Gu@2@AEBVPxMeshScale@2@AEBVPxTransform@2@@Z)' [\path\build\sdk_source_bin\PhysXCommon.vcxproj]
```

amoravanszky, 2019-12-27:
Yes, this dllimport/export dependency ugliness with foundation has been causing link problems on and off for a while now.  I'll check what we can do about it.

PierreTerdiman, 2021-04-09:
Thank you, this has been fixed in our trunks.

closed 2021-04-09 by PierreTerdiman